aws/amazon-freertos #99: Is there an existing toolchain for working in PlatformIO on VSCode?

wegunterjr, 2018-11-16:
I see the toolchain setup for command line, but would be awesome to see if there are instructions on how to use it in platformio on VSCode.

dcgaws, 2018-11-16:
Hi, do you mind sharing more about your target environment, including host-side OS (PC, Linux, or Mac?) and IoT dev board? 

I confess that I'm new to PlatformIO, although I'm already a big fan of VSCode. Anyway, any additional details you can share would help us prioritize. Thank you.

wegunterjr, 2018-11-16:
Windows 10 or Linux, ESP32 Wrover32 (i have a custom JSON file for the
board that includes the PSRAM)

I am building now with the PlatformIO (it has a lot of built in capabliity
that I am not using yet in my early dev).   It includes a certain version
of the esp-idf as a platform, but could also include Amazon FreeRTOS as
well.

abhidixi11, 2018-11-21:
Thanks for your input wegunterjr@. We'll keep that in mind when we prioritize.

closed 2018-11-21 by abhidixi11