aws/amazon-freertos #138: STM32L475 Sensors read 0.00

prothorin, 2018-12-05:
Hello

How to enable STM32L475 discovery board sensors to send via mqtt or greengrass?

SENSOR_IO_Init();
BSP_TSENSOR_Init();
BSP_HSENSOR_Init();
.
.
float temp;
temp = BSP_TSENSOR_ReadTemp();
( void ) snprintf( cDataBuffer, echoMAX_DATA_LENGTH,"Temp:  %2.f",temp );

But on mqtt topic:

Temp: 0.00

Thanks

pvyawaha, 2018-12-07:
Hello, It seems you are using the sensor library provided by ST in your application. We will consider adding sensor libraries in future development.

Can you check if the sensor is initialized properly and you can read the temperature in the variable ?

LMESTM, 2018-12-07:
@prothorin @pvyawaha 
My 2 cents ...
I haven't tested this particular code, but similar issues have already been reported before when using GCC tollchain and trying to printf or sprintf float . So here is something you can check:

Check your linker flags (project properties > C/C++ Build > Setings > Tool Settings (TAB) > MCU GCC Linker > Miscellaneous > Linker flags), 
if you are using nanolib “-specs=nano.specs” you need to add “-u _printf_float” to enable float printf.

Hints found here & here:
http://www.openstm32.org/forumthread2108
http://www.openstm32.org/forumthread954

Hoe this helps

prothorin, 2018-12-07:
Thanks now working,  sending sensors(humidity,temperature,magnetomete,gyroscope,accelerometer,barometer) data over mqtt to node-red.

closed 2018-12-07 by pvyawaha