aws/amazon-freertos #143: Greengrass demo on ESP32 returns eMQTTClientGotDisconnected

fast4break, 2018-12-06:
Run AWS FreeRTOS on a esp32 board,  the greengrass demo can not pass.  Log:
-------------------------------------
44 2468 [IoT_GGD] MQTT Agent Publishing Message start.
47 2468 [IoT_GGD] Sending command Event Type = 5
51 2471 [IoT_GGD] Woken by a notification message 
56 2471 [IoT_GGD] mqttNOTIFICATION_CODE  ==  34 (eMQTTClientGotDisconnected)
57 2471 [IoT_GGD] MQTT_AGENT_Publish command returns failure.

59 3621 [IoT_GGD] MQTT Agent Publishing Message start.
62 3621 [IoT_GGD] Sending command Event Type = 5
66 3623 [IoT_GGD] Woken by a notification message 
71 3623 [IoT_GGD] mqttNOTIFICATION_CODE  ==  12 (eMQTTPUBCouldNotBeSent)
72 3623 [IoT_GGD] MQTT_AGENT_Publish command returns failure.

74 4773 [IoT_GGD] MQTT Agent Publishing Message start.
77 4773 [IoT_GGD] Sending command Event Type = 5
81 4775 [IoT_GGD] Woken by a notification message 
86 4775 [IoT_GGD] mqttNOTIFICATION_CODE  ==  12(eMQTTPUBCouldNotBeSent) 
87 4775 [IoT_GGD] MQTT_AGENT_Publish command returns failure.

119 8229 [IoT_GGD] Disconnecting from broker......
122 8229 [IoT_GGD] Sending command Event Type = 2
126 8231 [IoT_GGD] Woken by a notification message 
131 8231 [IoT_GGD] mqttNOTIFICATION_CODE  ==  8(eMQTTDISCONNSent)
132 8231 [IoT_GGD] Disconnected from the broker.
133 8231 [IoT_GGD] Deleted Client.
134 8231 [IoT_GGD] Heap low watermark: 95892. Stack high watermark: 216.
135 8231 [IoT_GGD] ----Demo finished----
-------------------------------------

After debugging code, seems like the reason is that function SOCKETS_Connect is not impelmented for ESP32. Is this the reason of above problem?  Btw, several other demos without requiring GGC run ok. 
Thanks.

fast4break, 2018-12-07:
Run AWS FreeRTOS on a esp32 board, the greengrass demo can not pass. Log:
---------------------------->
44 2468 [IoT_GGD] MQTT Agent Publishing Message start.
47 2468 [IoT_GGD] Sending command Event Type = 5
51 2471 [IoT_GGD] Woken by a notification message
56 2471 [IoT_GGD] mqttNOTIFICATION_CODE == 34 (eMQTTClientGotDisconnected)
57 2471 [IoT_GGD] MQTT_AGENT_Publish command returns failure.

59 3621 [IoT_GGD] MQTT Agent Publishing Message start.
62 3621 [IoT_GGD] Sending command Event Type = 5
66 3623 [IoT_GGD] Woken by a notification message
71 3623 [IoT_GGD] mqttNOTIFICATION_CODE == 12 (eMQTTPUBCouldNotBeSent)
72 3623 [IoT_GGD] MQTT_AGENT_Publish command returns failure.

74 4773 [IoT_GGD] MQTT Agent Publishing Message start.
77 4773 [IoT_GGD] Sending command Event Type = 5
81 4775 [IoT_GGD] Woken by a notification message
86 4775 [IoT_GGD] mqttNOTIFICATION_CODE == 12(eMQTTPUBCouldNotBeSent)
87 4775 [IoT_GGD] MQTT_AGENT_Publish command returns failure.

119 8229 [IoT_GGD] Disconnecting from broker......
122 8229 [IoT_GGD] Sending command Event Type = 2
126 8231 [IoT_GGD] Woken by a notification message
131 8231 [IoT_GGD] mqttNOTIFICATION_CODE == 8(eMQTTDISCONNSent)
132 8231 [IoT_GGD] Disconnected from the broker.
133 8231 [IoT_GGD] Deleted Client.
134 8231 [IoT_GGD] Heap low watermark: 95892. Stack high watermark: 216.
135 8231 [IoT_GGD] ----Demo finished----
<-------------------------------------
After debugging code, seems like the reason is that function SOCKETS_Connect is not impelmented for ESP32. Is this the reason of above problem? Btw, several other demos without requiring GGC run ok.
Thanks.

gordonwang0, 2018-12-12:
Hi @fast4break,

Apologies for the delayed response.  I have seen something similar if the GG core crashes (i.e. it can be discovered but connection fails).  Can you ensure that your GG core is running?

fast4break, 2018-12-12:
GG core is running since I can recieve messages from HelloWorld_Publisher(Module 4 of DG). And ESP32 board can get correct IP address of the GG core whenever I change the GG core address(I tried different subnetworks). Not sure if GG core's CA is the problem, I just assume the example code will do things right. Can figure out what I am missing here.

gordonwang0, 2018-12-13:
Can you also check that you have configured your [GG subscriptions](https://docs.aws.amazon.com/greengrass/latest/developerguide/config_subs.html) and [AWS IoT policy](https://docs.aws.amazon.com/iot/latest/developerguide/iot-policies.html)?

Are you able to connect to this GG core using another client?

fast4break, 2018-12-14:
test1:
IoT Cloud<------>GG core on Pi<------->publisher/subscriber on laptop
=>Pub and sub can talk with each other through GG core

test2:
IoT Cloud<------>esp32
=>esp32 can send and recieve echo message to IoT Cloud

test3:
IoT Cloud<------>GG core on Pi<------->esp32
=>esp32 can find the correct IP address, but always recieve MQTT Disconnect error code(34) then error code 12 after invoking function prvSendCommandToMQTTTask.
___________________________
GG Subscribption:
          esp32-wrover -----------------> Local Shadow Service 
          Local Shadow Service -------------------> IoT Cloud
__________________________
Policy for esp32:
    {         "Effect": "Allow",          "Action": "iot:Connect",          "Resource": "arn:aws:iot:us-east-1:xxxxxxxx:*"    },
    {      "Effect": "Allow",      "Action": "iot:Publish",      "Resource": "arn:aws:iot:us-east-1:xxxxxxxx:*"    },
    {      "Effect": "Allow",      "Action": "iot:Subscribe",      "Resource": "arn:aws:iot:us-east-1:xxxxxxxx:*"    },
    {      "Effect": "Allow",      "Action": "iot:Receive",      "Resource": "arn:aws:iot:us-east-1:xxxxxxxx:*"    },
    {      "Effect": "Allow",      "Action": "greengrass:*",      "Resource": "arn:aws:iot:us-east-1:xxxxxxxx:*"    }

sarenameas, 2018-12-21:
Hello fast4break, 

1. To re-iterate what gordonwang0 said, please make sure that the Greengrass core has not crashed. Inside of the {...}/greengrass folder on my core, I know it has crashed when I find a non-empty "crash.log". Just to make sure we can go ahead and restart the core:
```
[ec2-user@ip-XXX-XXX-XXX-XXX greengrass]$ sudo ./greengrassd restart
Stopping greengrass daemon of PID: 14508
Waiting........
Stopped greengrass daemon, exiting with success
Setting up greengrass daemon
Validating execution environment
Found cgroup subsystem: cpuset
Found cgroup subsystem: cpu
Found cgroup subsystem: cpuacct
Found cgroup subsystem: blkio
Found cgroup subsystem: memory
Found cgroup subsystem: devices
Found cgroup subsystem: freezer
Found cgroup subsystem: net_cls
Found cgroup subsystem: perf_event
Found cgroup subsystem: net_prio
Found cgroup subsystem: hugetlb
Found cgroup subsystem: pids

Starting greengrass daemon.....
Greengrass daemon started with PID: 3503
[ec2-user@ip-XXX-XXX-XXX-XXX greengrass]$ ps -a
  PID TTY          TIME CMD
 3503 pts/0    00:00:00 daemon
 3598 pts/0    00:00:00 ps
```

2. In {...}/greengrass/configuration/config.json verify the contents are the expected certificate and iotHost to connect to the AWS IoT core:
```
[ec2-user@ip-XXX-XXX-XXX-XXX greengrass]$ cd configuration/
[ec2-user@ip-XXX-XXX-XXX-XXX configuration]$ ls
certs  config.json  group
[ec2-user@ip-1XXX-XXX-XXX-XXX configuration]$ cat config.json
{
    "coreThing": {
        "caPath": "root-certificate.pem",
        "certPath": "XXXX-certificate.pem.crt.txt",
        "keyPath": "XXXX-private.pem.key",
        "thingArn": "Greengrass core ARN",
        "iotHost": "YOUR AWS IoT Core endpoint",
        "ggHost": "greengrass.iot.us-west-2.amazonaws.com"
    },
    "runtime": {
        "cgroup": {
            "useSystemd": "yes"
        }
    }
}
```

2. In your Greengrass group in AWS IoT core please verify: 
- That your device is added to the list of devices
[image]
- That you have allowed subscriptions to either all topics (#) or the "freertos/demos/echo" topic. I have subscribed to all topics.
[image]

3. Make sure your have the proper permissions on resources in your thing's certificate's policy:
```json
{
    "Version": "2012-10-17",
    "Statement": [
    {
        "Effect": "Allow",
        "Action": "iot:Connect",
        "Resource":"arn:aws:iot:<aws-region>:<aws-account-id>:*"
	}, 
[... 155 lines cut ...]
9 2489 [IoT_GGD] Socket closed.
10 2489 [IoT_GGD] Stack high watermark for discovery helper task: 2740.
I (25110) PKCS11: Initializing SPIFFS
I (25120) PKCS11: Initializing SPIFFS
I (25220) PKCS11: Initializing SPIFFS
I (25220) PKCS11: Initializing SPIFFS
I (25220) PKCS11: Initializing SPIFFS
I (25960) PKCS11: Initializing SPIFFS
11 2760 [IoT_GGD] About to close socket.
12 2762 [IoT_GGD] Socket closed.
13 2762 [IoT_GGD] Stack high watermark for discovery helper task: 116.
14 2762 [IoT_GGD] Greengrass device discovered.
15 2762 [IoT_GGD] Establishing MQTT communication to Greengrass...
I (27750) PKCS11: Initializing SPIFFS
I (27750) PKCS11: Initializing SPIFFS
I (27860) PKCS11: Initializing SPIFFS
I (27860) PKCS11: Initializing SPIFFS
I (27860) PKCS11: Initializing SPIFFS
I (28660) PKCS11: Initializing SPIFFS
16 3484 [IoT_GGD] Disconnecting from broker.
17 3486 [IoT_GGD] Disconnected from the broker.
18 3486 [IoT_GGD] Deleted Client.
19 3486 [IoT_GGD] Heap low watermark: 109640. Stack high watermark: 116.
20 3486 [IoT_GGD] ----Demo finished----
```

closed 2018-12-27 by dcgaws