aws/amazon-freertos #158: Update ESP32 App Offset to land in ota_0 partition

nickrobinson, 2018-12-15:
I was working on getting the demo app working when I ran into the following error:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6040
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:15784
entry 0x400792e8
I (30) boot: ESP-IDF v3.1-dev-322-gf307f41-dirty 2nd stage bootloader
I (30) boot: compile time 11:03:17
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (66) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (74) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (81) boot:  3 ota_0            OTA app          00 10 00020000 00100000
I (89) boot:  4 ota_1            OTA app          00 11 00120000 00100000
I (96) boot:  5 storage          Unknown data     01 82 00220000 00010000
I (104) boot: End of partition table
E (108) boot: ota data partition invalid and no factory, will try all partitions
E (116) esp_image: image at 0x20000 has invalid magic byte
E (122) boot: OTA app partition slot 0 is not bootable
E (128) esp_image: image at 0x120000 has invalid magic byte
W (134) esp_image: image at 0x120000 has invalid SPI mode 255
W (141) esp_image: image at 0x120000 has invalid SPI size 15
E (147) boot: OTA app partition slot 1 is not bootable
E (153) boot: No bootable app partitions in the partition table
```

After running ```make size``` I got the following:

```
make size
Total sizes:
 DRAM .data size:    9232 bytes
 DRAM .bss  size:   58240 bytes
Used static DRAM:   67472 bytes ( 113264 available, 37.3% used)
Used static IRAM:   58210 bytes (  72862 available, 44.4% used)
      Flash code:  452098 bytes
    Flash rodata:   76284 bytes
Total image size:~ 595824 bytes (.bin may be padded larger)
```

By default the application is getting written to the nvs partition and overrunning the available space in that partition. I would suggest moving the default value in the line below to 0x20000. 

https://github.com/aws/amazon-freertos/blob/0743fb5bc274cfaedca02fa46d4cd87094a82adf/demos/espressif/esp32_devkitc_esp_wrover_kit/make/sdkconfig.defaults#L8

nickrobinson, 2018-12-15:
Fix in https://github.com/aws/amazon-freertos/pull/159

sarenameas, 2018-12-18:
Thanks again for your fix. We typically use the sdkconfig file already in the /make folder and therefore the sdkconfig.default is ignored. sdkconfig is in source control for our automatic build system, which cannot run "make menuconfig" to configure the serial port. I do see the benefit of making sure the sdkconfig.default is up to date and accurate. We will re-evaluate the current sdkconfig.default files and update them.

closed 2018-12-21 by sarenameas