aws/amazon-freertos #168: xSemaphoreGiveFromISR should fail when used with mutex

E-Brons, 2018-12-19:
Due to mutex inheritance feature, every TCB holds the amount of mutexes taken: pxTCB->uxMutexesHeld.
It shall never be allowed to release a mutex from ISR context. Currently such a faulty usage will 100% cause failures down the way - but the API returns pdPASS (success value).

Use case 1:
When mutex released from another context than taken - it will underflow one TCB's uxMutexesHeld and eventaully overflow the other - and anyway corrupt the value.

Use case 2:
A common semaphore usage scheme for peripheral drivers is take semaphor by task and give by ISR (for an example see https://www.freertos.org/Embedded-RTOS-Binary-Semaphores.html).

IMHO, at least use case 2, which is easily identified and commonly used - should be guarded by OS (return fail value). Fixing use case 1 will be nice to have too.

RichardBarry, 2018-12-19:
Hi, thanks for your post.

It is recommended to do development work with [configASSERT()](https://www.freertos.org/a00110.html#configASSERT) defined.  That will catch and alert the developer to both use cases you highlight in your post.  Once you code is executing without any assertions failing you can remove the configASSERT() definition in order to minimize the executable size and run time overhead.

Scenario 1 is caught by this assertion: https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS/tasks.c#L3970

Scenario 2 is caught by this assertion: https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS/queue.c#L1095

Note scenario 2 is fine with binary semaphores, and in the most common use cases a task notification provides a faster and lighter weight alternative.

E-Brons, 2018-12-27:
From now on I work with FreeRTOS API only with a proper configASSERT().

Thanks.

closed 2018-12-27 by E-Brons