aws/amazon-freertos #176: FreeRTOS Windows Simulator demo issue

prvthkr, 2018-12-27:
Hi,

I'm testing the demo for the [FreeRTOS Windows Simulator](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_windows.html)(Connect_to_AWS_IoT_-_Windows) and I got below log:

```
12 751 [MQTTEcho] MQTT echo attempting to connect to xxxxxxxxxxxxx7-ats.iot.ap-south-1.amazonaws.com.
13 15752 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
14 15752 [MQTTEcho] MQTT echo test could not connect to broker.
15 15752 [MQTTEcho] MQTT echo demo finished.
16 15752 [MQTTEcho] ----Demo finished----
```

```
The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which should be
defined in FreeRTOSConfig.h Attempting to open interface number 0.
```
Already defined to '0' IN FreeRTOSConfig.h file.

Can anybody has any idea about this?

dcgaws, 2018-12-27:
Hello @prvthkr, interface 0 is probably not the one you want. Instead, you want the interface index enumerated by WinPCap that corresponds to the wired/Ethernet connection that has internet access. 

For an example, please see the Configure Your Network Interface section of https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_windows.html. In the sample output that we pasted into that documentation, the desired interface was index 1.

prvthkr, 2018-12-28:
Hi @dcgaws,

Thank you for your inputs.
I configured as per your suggestion but still, I'm getting the same error, below is the snip:
```
0 0 [None] FreeRTOS_IPInit
1 0 [None] vTaskStartScheduler

The following network interfaces are available:

Interface 1 - rpcap://\Device\NPF_{xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx}
              (Network adapter 'Realtek Ethernet Controller' on local host)

Interface 2 - rpcap://\Device\NPF_{xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx}
              (Network adapter 'VMware Virtual Ethernet Adapter' on local host)

Interface 3 - rpcap://\Device\NPF_{xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx}
              (Network adapter 'VMware Virtual Ethernet Adapter' on local host)

The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which
should be defined in FreeRTOSConfig.h
Attempting to open interface number 1.
2 162 [IP-task] vDHCPProcess: offer 192.168.5.188
3 202 [IP-task] vDHCPProcess: offer 192.168.5.188
4 202 [IP-task] Write code signing certificate...
5 203 [IP-task] Write device private key...
6 225 [IP-task] Key provisioning done...
7 225 [IP-task] Creating MQTT Echo Task...
8 225 [IP-task]

IP Address: 192.168.5.188
9 225 [IP-task] Subnet Mask: 255.255.255.0
10 225 [IP-task] Gateway Address: 192.168.5.1
11 225 [IP-task] DNS Server Address: 172.28.80.16

12 225 [MQTTEcho] MQTT echo attempting to connect to xxxxxxxxxxxxxx-ats.iot.ap-south-1.amazonaws.com.
```

dcgaws, 2018-12-28:
@prvthkr, the next steps for troubleshooting depend on how far the device gets in its attempt to connect to the AWS IoT DNS endpoint shown in the last line of your logging trace above. Namely:

1. Does the device get a valid IP address from the DNS server for the AWS IoT endpoint? If not, there's either a DHCP issue or routing problem between the chosen interface ("1") and the internet.
2. Is the the device able to establish a TCP connection, but then the TLS negotiation fails? If so, it could be that the device doesn't trust the server certificate, the server doesn't trust the device certificate, or that the device is trying to negotiated a ciphersuite that the server doesn't like. 
3. Does the TLS negotiation succeed, but then immediately gets disconnected? If so, the device "Thing" policy in the AWS IoT registry is misconfigured.

mradulan, 2019-01-08:
Hi @prvthkr, are you still getting the error?

prvthkr, 2019-01-08:
Hi @mradulan, yes still I'm getting the error. Detail information I'll post in a few hours.

abhidixi11, 2019-01-15:
Hi @prvthkr did you get chance to follow the next steps listed above and collect the information ?  Maybe you can get packet capture from the windows machine, which will help to find out what is going on. Thanks.

varshakiragi, 2019-01-17:
I am very new to AWS usage and Even I am facing the same issue.. 
Below is output:
The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which
should be defined in FreeRTOSConfig.h
Attempting to open interface number 1.
2 75 [IP-task] vDHCPProcess: offer 192.168.5.101
3 115 [IP-task] vDHCPProcess: offer 192.168.5.101
4 115 [IP-task] Write code signing certificate...
5 115 [IP-task] Write device private key...
6 121 [IP-task] Key provisioning done...
7 121 [IP-task] Creating MQTT Echo Task...
8 121 [IP-task]

IP Address: **192.168.5.101**
9 121 [IP-task] Subnet Mask: 255.255.255.0
10 121 [IP-task] Gateway Address: 192.168.5.1
11 121 [IP-task] DNS Server Address: 172.28.80.16

12 121 [MQTTEcho] MQTT echo attempting to connect to a2r4zp84hezzja-ats.iot.us-west-2.amazonaws.com.
13 175 [IP-task] vDHCPProcess: offer 192.168.5.101
14 175 [IP-task]

1. Here, actual IP address of my system is 192.168.5.224 but above output shows  **vDHCPProcess: offer 192.168.5.101**... Is this expected? (both IP address are different?).
2. I found in above post, Is the the device able to establish a TCP connection, but then the TLS negotiation fails? Here **how to verify TCP connection is fine or TLS negotiation fails?**
3. How to get **packet capture** from the windows machine?

Also, 
I tried execute following cmd:
$**aws iot describe-endpoint**
but I am getting following error:
SSL validation failed for https://iot.us-west-2.amazonaws.com/endpoint [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)

**Is there any relation between these 2 issues...?**

htibosch, 2019-01-17:
> Here, actual IP address of my system is 192.168.5.224 but above
> output shows vDHCPProcess: offer 192.168.5.101...
> Is this expected? (both IP address are different?).

The demo projects require that you define a default IP-address for your device in FreeRTOSConfig.h as `configIP_ADDR[0..3]`.
Apparently you have DHCP enabled in your FreeRTOSIPConfig.h with `ipconfigUSE_DHCP`. DHCP is free to assign any new address to your device.
In case there is **no DHCP**, the demo will use the configured default IP-address (and netmask, DNS etc.)
In real life, I think it is a good custom to remember the last used network settings, and use those in case DHCP is non-functional

htibosch, 2019-01-17:
> 3 How to get packet capture from the windows machine?

Can't you use a program like WireShark and choose the LAN adapter?

varshakiragi, 2019-01-18:
[AWS.zip](https://github.com/aws/amazon-freertos/files/2771593/AWS.zip)
Attaching captured wireshark files.. Request you to go through it..
It will be really helpful if you can analyze and let me know what is going wrong.

Also, please have look at below error for **$aws iot describe-endpoint**
SSL validation failed for https://iot.us-west-2.amazonaws.com/endpoint [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)

aggarg, 2019-01-18:
Since the demo is failing in the connect to AWS IoT step, this might be an issue with your policy setup. Would you please try our Quick Onboarding workflow to ensure that everything is setup correctly? The steps to use onboarding workflow are detailed in this document: [OnBoardingWorkflow.pdf](https://github.com/aws/amazon-freertos/files/2774186/OnBoardingWorkflow.pdf)
 
Regarding your other questions:
>> Here, actual IP address of my system is 192.168.5.224 but above output shows vDHCPProcess: offer 192.168.5.101... Is this expected? (both IP address are different?).

Yes this is expected as Amazon FreeRTOS creates a separate TCP/IP stack.

>>Also, please have look at below error for $aws iot describe-endpoint
SSL validation failed for https://iot.us-west-2.amazonaws.com/endpoint [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)

This seems like an AWS CLI issue and the two issues are probably not related. Are you using the latest version of AWS CLI? If not, would you please update it. These threads provide additional information about the same issue:
- https://github.com/aws/aws-cli/issues/3504
- https://github.com/aws/aws-cli/issues/3009

Also, when you capture packets using Wireshark, you can use a capture filter to ensure that only relevant packets are captured: https://wiki.wireshark.org/CaptureFilters. Assuming that the IP assigned by DHCP is 192.168.5.101, you can use the following capture filter: ```host 192.168.5.101```

Thanks.

varshakiragi, 2019-01-22:
> Since the demo is failing in the connect to AWS IoT step, this might be an issue with your policy setup. Would you please try our Quick Onboarding workflow to ensure that everything is setup [...]

I made this setup again following instructions in the pdf that u have provided. But still facing same issue.

I have a question, can we use ****Wifi** for windows simulator example project**?

RichardBarry, 2019-01-22:
If it helps, I recently got one of these to connect the Ethernet on my laptop to the Wi-Fi in my house - it allows me to use the Windows Simulator project when I am out of reach of a wired Ethernet connection.  https://www.amazon.com/gp/product/B014SK2H6W/

varshakiragi, 2019-01-23:
> Would you please share your complete logs to help us better understand the issue? Also, would you like to have a quick debug session to debug this issue?

[AWS.zip](https://github.com/aws/amazon-freertos/files/2785724/AWS.zip)

[23-01-2019.zip](https://github.com/aws/amazon-freertos/files/2785731/23-01-2019.zip)
[image]

Debug session would be better option.
I am suspecting some proxy issue at my side. Can u please tell me what will be URL or remote IP address that will be involved in TCP connection for this project?

aggarg, 2019-01-24:
Looking at the Wireshark logs you provided, it seems that 2 DHCP servers are responding with the same IP assignment:
[image]

Would you please check why do you have 2 DHCP servers on the same subnet:  ```192.168.5.21``` and ```192.168.5.22```.

>> I am suspecting some proxy issue at my side. Can u please tell me what will be URL or remote IP address that will be involved in TCP connection for this project?

It is printed in your logs: ```a2r4zp84hezzja-ats.iot.us-west-2.amazonaws.com```

Thanks.

aggarg, 2019-01-28:
Hi @varshakiragi ,

Did you get a chance to look into it? 

Thanks.

mradulan, 2019-02-11:
closing this issue since no response received in last 14 days. Please reopen if issue persists.

closed 2019-02-11 by mradulan

snahmad, 2019-11-09:
Does this simulator works with WiFi only. No ethernet