aws/amazon-freertos #183: How to resolve "NetInterface: Dropping packet"

JSchaenzle, 2018-12-28:
I'm working on a project using the supported ESP32 hardware which sends and receives data using MQTT. When running my app with the monitor open I frequently see this message:
`NetInterface: Dropping packet`
This is happening more than once per second. What causes packets to be dropped so frequently and how can I prevent it?

jrr, 2018-12-31:
Looks like it comes from here:
https://github.com/aws/amazon-freertos/blob/12ea022d0f52df211111a79e0aca1ca7b04639d6/lib/FreeRTOS-Plus-TCP/source/portable/NetworkInterface/esp32/NetworkInterface.c#L100-L101

Checking out the return conditions of `eConsiderFrameForProcessing`:
https://github.com/aws/amazon-freertos/blob/5497db65a33503a77bb71850ea80eef3b2497b0e/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_IP.c#L1244-L1246

This seems like a normal case of discarding packets not meant for the device. Does it merit logging?

mradulan, 2019-01-09:
Hi @JSchaenzle , it looks like the device is receiving packets which are not targeted towards it, so the network interface is discarding those. Would you be able to provide a log of the output?

mahavirj, 2019-01-10:
@JSchaenzle 
> When running my app with the monitor open I frequently see this message:
NetInterface: Dropping packet
This is happening more than once per second. What causes packets to be dropped so frequently 

Most likely `multicast` service running in local network can be one of the reason.

@jrr 
> This seems like a normal case of discarding packets not meant for the device. Does it merit logging?

Yes normal case of discarding packets. Regarding logging, it is by default disabled (only gets enabled if debug logs are explicitly enabled).

yuhui-zheng, 2019-01-30:
This issue has been stale for more than 2 weeks. Please contact us if there's further question. Closing the issue for now.

closed 2019-01-30 by yuhui-zheng