aws/amazon-freertos #194: The new FreeRTOS TAG v1.4.6 didn't use header file "mpu_prototypes.h"

votuananhs, 2019-01-05:
I have attempted to integrate MPU features in my project. But I got an building issue following message error:
`amazon-freertos\lib\FreeRTOS\portable\Common\mpu_wrappers.c",45  Fatal error[Pe1696]: cannot open source file "mpu_prototypes.h" `

In my suggestion, we can remove this header file in mpu_wrappers.c files. Because we just include the mpu_wrappers.h file in application. Don't include list of FreeRTOS's header files:

/* Scheduler includes. */
#include "FreeRTOS.h"
#include "task.h"
#include "queue.h"
#include "timers.h"
#include "event_groups.h"
#include "stream_buffer.h"
#include "mpu_prototypes.h"

RichardBarry, 2019-01-06:
Hi - I just had a look at the v1.4.6 tag and confirmed mpu_prototypes.h does indeed appear to have been omitted.  It will get added, until which time, you can obtain the file from here: https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS/Source/include/

The order in which the mpu header files get included is important to ensure API functions get remapped to their MPU_ equivalents in some files and not others, so I don't recommend editing the kernel code in the way you suggest.

votuananhs, 2019-01-08:
hi Richard Barry, 
I got this suggestion, because I have just included **mpu_wrappers.h** header in my application and it will keep same naming API in MPU feature support with case doesn't define macro **portUSING_MPU_WRAPPERS** in **FreeRTOSConfig.h**
I have tried to build with this solution. And didn't see any error for now.

mradulan, 2019-01-08:
Hi @votuananhs , thank you for pointing out. We will be adding this to the repo. Meanwhile, I am closing this issue.

closed 2019-01-08 by mradulan