aws/amazon-freertos #202: Wifi connection unsuccessful sometimes on ESP32 device.

aehrath, 2019-01-09:
I am running into an issue, where on an ESP32S device (similar to DevkitC) the device will fail to connect to my wifi 50% of the time or so. 
I am running the AWS shadow sample code.
Here is the failure:
I (207) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (217) wifi: mode : sta (30:ae:a4:1c:98:60)
I (217) WIFI: SYSTEM_EVENT_STA_START
I (2627) WIFI: SYSTEM_EVENT_STA_DISCONNECTED
1 253 [main] WiFi failed to connect to AP.

mradulan, 2019-01-10:
The network connection via VM is generally unreliable. Although can you check which configuration your `Network Adapter` is using (NAT or Bridged)? Also were you able to connect to the Wifi on your Windows host machine?

aehrath, 2019-01-10:
What you said does not make sense. The wifi connection between the ESP32 and my wifi network has nothing to do with the host machine. It connects directly to the wifi, circumventing the host. The only thing the host does, is receive log messages and send commands to the device. The ESP32 connects fine to my network about 50% of the time or so. (more or less random)

mradulan, 2019-01-10:
There are some steps which will help debug more:
1. Can you use WIFI_Scan method defined in [this ](https://github.com/aws/amazon-freertos/blob/master/lib/include/aws_wifi.h#L126It)header to check the signal strength?
2. What is the model of your AP?
3. There is a similar [issue](https://github.com/aws/amazon-freertos/issues/57) which might be useful.

aehrath, 2019-01-16:
AP is netgear nighthawk. signal strength is between -65 to -68.

aggarg, 2019-01-16:
Thank you for sharing the information. Would you please try the following steps to help us better understand the issue:
- Please try using a different network with better signal strength (-55 or better) for reliable connection.
- If that does not solve the problem, would you please take the code in this branch and share the logs: https://github.com/aws/amazon-freertos/tree/merge/ESPpatch

The above mentioned branch contains the latest ESP-IDF with enhanced logging which will help us debug the issue.

Thanks.

yuhui-zheng, 2019-02-05:
This thread seems in inactive for ~three weeks. Please reach out again and reference this ticket, if things are still not working for you. Closing issue...

closed 2019-02-05 by yuhui-zheng