aws/amazon-freertos #203: Shadow_ClientConnect fails sometimes on ESP32 device.

aehrath, 2019-01-09:
I am running into an issue on an ESP32S device (similar to DevkitC) the shadow demo fails to connect the shadow client with error code 2 about 20% of the time.
Below is the log:
6 345 [main] Key provisioning done...
I (4567) wifi: pm start, type:0

7 848 [MainDemoTask] Shadow_ClientConnect unsuccessful, returned 2.
8 849 [MainDemoTask] Failed to initialize Shadow client.

abhidixi11, 2019-01-15:
Hello aehrath ! 
Can you please enable MQTT debug logs and see if you can get any more information? Thanks

mradulan, 2019-02-10:
closing issue since no response received. Please reopen if the issue persists.

closed 2019-02-10 by mradulan

Utsav22G, 2019-07-12:
> Hello aehrath !
> Can you please enable MQTT debug logs and see if you can get any more information? Thanks

How do I enable the debug logs? Is this specific to my IDE? Is there a C file I need to edit??

For context, I'm using NXP LPC54018 with MCUXpresso, and am getting the same error.