aws/amazon-freertos #207: Missing support for WPA2 enterprise

w-joost, 2019-01-10:
Can you please add support for WPA2 enterprise? I need the option to authenticate using an username and password or a certificate to connect to our wlan.
At least the espressif esp32 sdk version 3.1.2 has working support for it. However, the AWS toolkit doesn't support setting an username or adding a key/certificate pair.
I tried to add WPA2 enterprise by myself to amazon-freertos, but it didn't work. Maybe the version of the esp32 sdk shipped with amazon-freertos is too old.

mradulan, 2019-01-14:
Hi @w-joost , to configure WPA2 enterprise, you have to follow the guidelines in https://github.com/espressif/esp-idf/tree/release/v3.1/examples/wifi/wpa2_enterprise. After configuring it, you need to invoke wpa2 configuration APIs before starting WiFi in the function `WIFI_ConnectAP`.
As you mentioned, esp-idf SDK bundled with Amazon FreeRTOS is older (v3.0) , we will be updating it to the latest soon.

w-joost, 2019-01-15:
Hello @mradulan , thank you for your help. I managed to connect to our WLAN using the espressif toolkit, but it doesn't work with the sdk bundled with amazon-freertos. I found three commits with corrections for WPA2 enterprise:

* https://github.com/espressif/esp-idf/commit/bc97b0f380210b97b7fd98c5f304af7e83c28b7f
* https://github.com/espressif/esp-idf/commit/1d5dc2b9ef55d42b935fe1b1ea4449962a025280
* https://github.com/espressif/esp-idf/commit/9ad19e8122930034da97b8e15d32d3a3124d23c0

At least these commits are missing in the shipped toolkit.

w-joost, 2019-01-16:
I was able to connect to our wlan using the branch merge/ESPpatch and the attached diff. However, I only tested password based wpa2 enterprise authentication.

[wifi-abstraction-wpa2-enterprise-new.diff.txt](https://github.com/aws/amazon-freertos/files/2763661/wifi-abstraction-wpa2-enterprise-new.diff.txt)

mradulan, 2019-01-16:
The branch merge/ESPpatch does have the latest version ESP-IDF. Thank you attaching the diff logs. The example for certificate based wpa2 enterprise authentication is over[ here](https://github.com/espressif/esp-idf/blob/release/v3.1/examples/wifi/wpa2_enterprise/main/wpa2_enterprise_main.c) .

yuhui-zheng, 2019-01-30:
Please let us know if you still have issue. Since this thread has been stale for 14 days, closing.

closed 2019-01-30 by yuhui-zheng