aws/amazon-freertos #213: Add preprocessor warnings to ensure device/Thing configuration

dcgaws, 2019-01-15:
Customer feedback:

'I have one suggestion about a problem that I encountered:

    static const char clientcredentialMQTT_BROKER_ENDPOINT[] =
        "Paste AWS IoT Broker endpoint here.";

and similar statements: it took me a while before I discovered that I had failed
to change the above declaration.

Wireshark showed a DNS request for a domain called:

    "Paste AWS IoT Broker endpoint here."

It would be nice if the compiler could give a #warning or an #error about this.
And if not, the handling code could issue a warning.'

sarenameas, 2019-03-14:
We could possibly make this into a macro, define it in some other c file that all the projects are building, then extern clientcredentialMQTT_BROKER_ENDPOINT. This is for a small blast radius as was done with the clientcredential keys.

dcgaws, 2019-03-18:
Consistent with #219 would be to turn it into a #define anyway. In other words, no matter static const. That way, the developer is free to implement it as a "GET" function and not have to worry (as much, anyway) about breaking callers. The next step could be for us to add some code that error-checks the resulting value at runtime - see if it's a NULL pointer or empty string for example.

abhidixi11, 2019-04-04:
I have created internal feature request to track this issue.

dcgaws, 2019-05-29:
Fixed per the commit above.

closed 2019-05-29 by dcgaws