aws/amazon-freertos #219: IOT_THING_NAME : Extention Compatiblity

sbertrand-witekio, 2019-01-15:
The current usage of the clientcredentialIOT_THING_NAME define imposes to have it being as a constant literal string.

Can AFR handle clientcredentialIOT_THING_NAME as a string variable so system can be configured dynamically based on the thing name ?

dcgaws, 2019-01-16:
Hello, a change we're considering is to implement a "get" and "set" abstraction for storage of configuration such as the above. As you point out, that would entail modifying library code such as Greengrass Discovery. 

I suspect that we would still want some kind of default behavior so that lab testing can easily be done with statically compiled values. So there would potentially still be the #defines, but only the respective "get" routine would reference it. 

Would such a solution serve your needs?

sbertrand-witekio, 2019-01-16:
Indeed it would. As for default value, I have mine as such :

extern char MyDeviceName[];

char MyDeviceName[14]="MyDefaultVal";

Then it get reported using the abstraction, like in pcApplicationHostnameHook;

dcgaws, 2019-02-26:
There appeared to be one location - in the Greengrass Discovery library - where our code had a hard dependency on the Thing name #define being a preprocessor string literal. I have fixed that with the above commit.

Now, you should be all clear to use the existing Thing name #define however you like (i.e. including defining it to a function that, for example, reads/allocates the Thing name on the first call and returns a pointer to it).

Can you please let me know if that doesn't work? Thank you.

closed 2019-02-26 by dcgaws