aws/amazon-freertos #222: esptool write_flash error for ESP32

SaturnR, 2019-01-17:
I followed this  [amazon tutorial](https://aws.amazon.com/blogs/apn/connect-microcontroller-based-devices-to-the-cloud-with-amazon-freertos-and-espressif-esp32/). Everything worked fine, but when it starts to flash firmware to ESP32 esptool fails with error:
`esptool write_flash: error: argument <address> <filename>: Address "toflash/aws_demos.bin" must be a number
` 
Then i just compiled with make and then tried flash with this:

`esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 toflash/bootloader.bin toflash/aws_demos.bin 0x8000 toflash/partitions_example.bin`

I am using esptool, which is located in `lib/third_party/mcu_vendor/espressif/esp-idf/components/esptool_py/esptool` directory.

Is it auto formatting bug, incompatibility or something else? 

Thank you!

gilbertgit, 2019-01-17:
I have the same issue. Please let me know if you figure something out.

mbruno-xmos, 2019-01-18:
Same here with the latest commit. It used to work.

xuelix, 2019-01-18:
For quick workaround, just change the permission of this file 
chmod 755 parttool.py

You should be able to flash the board.

gilbertgit, 2019-01-18:
That worked. Thank you!

abhidixi11, 2019-01-19:
I will commit py files with 755 mode in the next PR, so you should not face the problem next week.  I'm closing the thread since fix is in the flight.  Thanks for reporting.

closed 2019-01-19 by abhidixi11