aws/amazon-freertos #227: STM32 aws_demos reset after 1 hour - Bug or feature?

egoltzman, 2019-01-20:
I'm working with the latest version (1.4.6) of amazon-freertos demos on B-L475E-IOT01A Discovery kit and I see that after an hour of work the system resets and restart.
Why? Is this a bug? Or a featuer? if a feature then where can I contorl it, if at all?

abhidixi11, 2019-01-21:
Hello  egoltzman !  
Which demo are you running ?  Can you please post the log here ?
Thanks.

egoltzman, 2019-01-21:
Hello abhidixi11
Thank you for your mail.
The demo I'm running is vStartShadowDemoTasks
The log is attached, here it did not take an hour, only half an hour.
Thanks

0 521 [Tmr Svc] WiFi module initialized.
1 532 [Tmr Svc] Write code signing certificate...
2 554 [Tmr Svc] Write device private key...
3 1405 [Tmr Svc] Key provisioning done...
4 4571 [Tmr Svc] WiFi connected to AP XYZ.
5 4575 [Tmr Svc] IP Address acquired 192.168.0.32
6 4580 [Tmr Svc] WiFi firmware version is: C3.5.2.5.STM
7 4585 [Tmr Svc] WiFi firmware is up-to-date.
8 4589 [MainDemoTask] [Shadow 0] MQTT: Creation of dedicated MQTT client succeeded.
9 7416 [MainDemoTask] [Shadow 0] MQTT: Connect succeeded.
10 8430 [MainDemoTask] [Shadow 0] MQTT: Subscribe to accepted topic succeeded.
11 9446 [MainDemoTask] [Shadow 0] MQTT: Subscribe to rejected topic succeeded.
12 9458 [MainDemoTask] [Shadow 0] MQTT: Publish to operation topic succeeded.
13 11474 [MainDemoTask] [Shadow 0] MQTT: Unsubscribe from rejected topic succeeded.
14 12489 [MainDemoTask] [Shadow 0] MQTT: Subscribe to callback topic succeeded.
15 12496 [MainDemoTask] Shadow client initialized.
16 12501 [ShDemoUpdt] Performing Thing Shadow update.
17 13515 [ShDemoUpdt] [Shadow 0] MQTT: Subscribe to accepted topic succeeded.
18 14531 [ShDemoUpdt] [Shadow 0] MQTT: Subscribe to rejected topic succeeded.
19 14543 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
20 15552 [ShDemoUpdt] Successfully performed update.
21 15556 [ShDemoUpdt] Performing Thing Shadow update.
22 15561 [Shd-IOT-0] Shd-IOT-0 changing desired state.
23 15571 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
24 16579 [ShDemoUpdt] Successfully performed update.
25 16584 [ShDemoUpdt] Performing Thing Shadow update.
26 16588 [Shd-IOT-1] Shd-IOT-1 changing desired state.
27 16598 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
28 17602 [ShDemoUpdt] Successfully performed update.
29 17607 [ShDemoUpdt] Performing Thing Shadow update.
30 17612 [Shd-IOT-0] Shd-IOT-0 done changing desired state.
31 17622 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
32 18630 [ShDemoUpdt] Successfully performed update.
33 18635 [Shd-IOT-1] Shd-IOT-1 done changing desired state.
34 20561 [Shd-IOT-0] Shd-IOT-0 changing desired state.
35 20566 [ShDemoUpdt] Performing Thing Shadow update.
36 20576 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
37 21589 [MQTT] Successfully added new reported state to update queue.
38 21598 [ShDemoUpdt] Successfully performed update.
39 21603 [Shd-IOT-1] Shd-IOT-1 changing desired state.
40 21608 [ShDemoUpdt] Performing Thing Shadow update.
41 21613 [Shd-IOT-0] Shd-IOT-0 done changing desired state.
42 21623 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
43 22632 [ShDemoUpdt] Successfully performed update.
44 22637 [ShDemoUpdt] Performing Thing Shadow update.
45 22646 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
46 23652 [MQTT] Successfully added new reported state to update queue.
47 23667 [ShDemoUpdt] Successfully performed update.
48 23672 [ShDemoUpdt] Performing Thing Shadow update.
49 23677 [Shd-IOT-1] Shd-IOT-1 done changing desired state.
50 23687 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
51 24696 [ShDemoUpdt] Successfully performed update.
52 25561 [Shd-IOT-0] Shd-IOT-0 changing desired state.
53 25566 [ShDemoUpdt] Performing Thing Shadow update.
54 25575 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
55 26581 [MQTT] Successfully added new reported state to update queue.
56 26596 [Shd-IOT-1] Shd-IOT-1 changing desired state.
57 26601 [ShDemoUpdt] Successfully performed update.
58 26606 [ShDemoUpdt] Performing Thing Shadow update.
59 26611 [Shd-IOT-0] Shd-IOT-0 done changing desired state.
60 26621 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
61 27629 [ShDemoUpdt] Successfully performed update.
62 27634 [ShDemoUpdt] Performing Thing Shadow update.
63 27643 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
[... 915 lines cut ...]
979 330565 [Shd-IOT-0] Shd-IOT-0 done changing desired state.
980 330576 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
981 330583 [Shd-IOT-0] Shd-IOT-0 changing desired state.
982 331584 [ShDemoUpdt] Successfully performed update.
983 331589 [MQTT] Successfully added new reported state to update queue.
984 331598 [ShDemoUpdt] Performing Thing Shadow update.
985 331608 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
986 335615 [MQTT] Update queue full, deferring reported state update.
987 335631 [ShDemoUpdt] Successfully performed update.
988 335636 [ShDemoUpdt] Performing Thing Shadow update.
989 335646 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
990 336652 [MQTT] Successfully added new reported state to update queue.
991 336668 [ShDemoUpdt] Successfully performed update.
992 336673 [ShDemoUpdt] Performing Thing Shadow update.
993 336683 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
994 337689 [MQTT] Successfully added new reported state to update queue.
995 337705 [ShDemoUpdt] Successfully performed update.
996 337710 [ShDemoUpdt] Performing Thing Shadow update.
997 337715 [Shd-IOT-1] Shd-IOT-1 done changing desired state.
998 337725 [ShDemoUpdt] [Shadow 0] MQTT: Publish to operation topic succeeded.
999 337732 [Shd-IOT-1] Shd-IOT-1 changing desired state.
1000 338733 [ShDemoUpdt] Successfully performed update.
1001 338738 [ShDemoUpdt] Performing Thing Shadow update.
1002 338744 [Shd-IOT-0] Shd-IOT-0 done changing desired state.
1003 338754 [ShDemoUpdt] [Shadow 0] MQTT: Publish to oper

alexa-noxon, 2019-01-22:
Hi egoltzman,

I do not think this is a feature :)  

The logs you posted don't appear to show any error or mention the device restarting-- are these the full logs or were they truncated?  Does the device have any abnormal messages that print out just before it restarts?  Can you tell for sure that the device restarts (vs hanging somewhere without restarting?).  

At my desk I have not been able to replicate your issue (I've been running the shadow update demo for about an hour and half on the development board - I will keep it going for a while longer just in case though!). Have you modified the project in any other way?

Thanks,
Alexa

egoltzman, 2019-01-22:
Hi Alexa,

Me neither (regarding the feature)...

I'm using a clean unchanged code from the git, and indeed there is no error
messages in the log (this is the full log) that reflect the fact that the
board is reset after a while (0.5-1hr).

The good thing is that after the reset it start again and continue with its
job so if you just look at it over time you can miss the reset (the time
stamp should show that it was restart).

Can you send me your bin? maybe my build is different?

Thanks,

Eyal

alexa-noxon, 2019-01-22:
Hi Eyal,

I left the board running overnight, and I was able to see the resets you mentioned (the first probably happened shortly after I posted!). 

Thanks for sharing that about the watchdog-- it sounds like a good hunch and thank you for looking into it!  I can also do some investigating on my side an hopefully we can get this resolved.

Thanks for pointing this out,
Alexa

alexa-noxon, 2019-04-04:
Hi Eyal,

Apologies for the delay in getting back to you. 

After taking these steps, I am no longer able to see the issue:
1. Use ST-LINK Utility to update the firmware on the device (I have updated to V2.J33.M25)
1. Disable all power saving settings for USB on the PC that is connected to the device
1. Update System Workbench to the latest version before compiling (click Help->Check for Updates)

I hope this will take care of the problem for you and any others as well.

Best,
Alexa

abhidixi11, 2019-04-05:
Hello Eyal,
I'm closing the issue.  Please feel free to re-open if the above steps don't resolve the reset problem for you.
Thank you !

closed 2019-04-08 by alexa-noxon