aws/amazon-freertos #260: How to port custom target.

Joyadam2, 2019-01-29:
How to port custom target which does not has inbuilt wi-fi feature.
I have ported the FreeRTOS and it's working correctly.

Can anybody help me regarding this (using any Wi-Fi module like ESP8266 or ESP32)?

dcgaws, 2019-01-30:
Hello @Joyadam2, for reference, please take a look at our ESP32 network port here https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS-Plus-TCP/source/portable/NetworkInterface/esp32/NetworkInterface.c. 

More generally, porting documentation for Amazon FreeRTOS can be found here https://docs.aws.amazon.com/freertos/latest/userguide/porting-guide.html.

prvthkr, 2019-02-04:
Hello @dcgaws , Thank you for your valuable inputs.

>for reference, please take a look at our ESP32 network port here https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS-Plus-TCP/source/portable/NetworkInterface/esp32/NetworkInterface.c.

As per my understanding after analyses the code, above reference is using the in-built wifi functionality of ESP32 but my requirement is different like ESP32 interface with custom board using UART or any other protocol.

>More generally, porting documentation for Amazon FreeRTOS can be found here https://docs.aws.amazon.com/freertos/latest/userguide/porting-guide.html.

In above porting guide they have not specified which all the library files are required for projects.

E.g.: they have given that how to port and what to port but how to create the project using Keil, IAR, 
and Visual Studio etc. (Create project means what are the things require to add in the projects apart from ported drivers) they have not given (please give me the link if someone explained everything).

thanks.

Joyadam2, 2019-02-04:
Hello @dcgaws, Thank you for your valuable inputs.

>for reference, please take a look at our ESP32 network port here https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS-Plus-TCP/source/portable/NetworkInterface/esp32/NetworkInterface.c.

As per my understanding after analyses, the code, above reference is using the in-built wifi functionality of ESP32 but my requirement is different like ESP32 interface with the custom board using UART or any other protocol.

>More generally, porting documentation for Amazon FreeRTOS can be found here https://docs.aws.amazon.com/freertos/latest/userguide/porting-guide.html.

In the above porting guide, they have not specified which all the library files are required for projects.

E.g.: they have given that how to port and what to port but how to create the project using Keil, IAR,
and Visual Studio etc. (Create project means what are the things required to add in the projects apart from ported drivers) they have not given (please give me the link if someone explained everything).

Thanks.

dcgaws, 2019-02-04:
Hello @Joyadam2, it's true that we don't provide lists of project files separate from the respective build environment, or IDE, for each development board. One reason is that it's tough to keep such a list up to date. 

Another reason we don't keep a separate list of files is that, in my experience, it's easier to start with a working project and then modify it according to your needs. In the case of ESP32, the build environment for the Amazon FreeRTOS demo and test projects is ESP-IDF and Make. 

Regarding possible starting points for your project, essentially all of our demos and tests connect to Wi-Fi during startup. Could you, for example, start with our MQTT Hello World demo and confirm that you have a working configuration end-to-end using an ESP32 development board, local Wi-Fi, and your AWS IoT account? Then use that same project as the basis for your custom board, gradually adding the missing functionality, starting with the UART driver. 

Are you planning on using a UART driver from ESP-IDF? May I ask what Wi-Fi part you've selected?

Thank you.

mradulan, 2019-02-10:
Hi @Joyadam2 , ESP8266 uses UART to connect to the MCU and requires AT commands to be sent to setup the wifi. There is an example of it being connected to Arduino over [here](https://www.hackster.io/user16726/configuring-the-esp8266-using-an-arduino-0ab2e6). You will probably have to implement aws_wifi.c for your custom board using the [guide](https://docs.aws.amazon.com/freertos/latest/lib-ref/html2/wifi/index.html). We don't have any WiFi module interfaced over UART but the STM32's Wifi driver communicates over a SPI bus and that can be a good reference - https://github.com/aws/amazon-freertos/blob/master/lib/wifi/portable/st/stm32l475_discovery/aws_wifi.c

Hope this helps.

Joyadam2, 2019-02-13:
Hi @mradulan, Thank you for your help.

>ESP8266 uses UART to connect to the MCU and requires AT commands to be sent to setup the wifi. There is an example of it being connected to Arduino over [here](https://www.hackster.io/user16726/configuring-the-esp8266-using-an-arduino-0ab2e6).

I'm closing this issue because right now I'm trying to port ESP8266 as per above. So, if I'll face any issue again I will reopen.

Thanks.

closed 2019-02-13 by Joyadam2