aws/amazon-freertos #265: esp32: spi_intr calls esp_intr_disable from interrupt context triggering vPortAssertIfInISR

btashton, 2019-01-30:
**Describe the bug**
The function `spi_intr` from the included  esp-idf requires the changes espressif made to `vTaskEnterCritical` to be able to disable interrupts from the interrupt context.
https://github.com/aws/amazon-freertos/blob/2fbb44cf6ea82895486af42b617f90ca1736e0e0/lib/third_party/mcu_vendor/espressif/esp-idf/components/driver/spi_master.c#L551

See the backtrace.

**System information**
- ESP-WROVER-KIT V4.1
- Operating System: MacOS
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it): `v1.4.6-41-g9697cea5`
- Project: Custom Application

**Expected behavior**
It is expected that the spi_master should be able to disable the interrupt.

**Screenshots or console output**
```
0x4008cb6f: vPortAssertIfInISR at /Users/brennana/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

Backtrace: 0x4009039c:0x3ffc0530 0x4009058f:0x3ffc0550 0x4008cb6f:0x3ffc0570 0x4008e3d2:0x3ffc0590 0x40081b76:0x3ffc05b0 0x4008be82:0x3ffc05d0 0x40083682:0x3ffc0600 0x4008e415:0x00000000
0x4009039c: invoke_abort at /Users/brennana/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x4009058f: abort at /Users/brennana/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x4008cb6f: vPortAssertIfInISR at /Users/brennana/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

0x4008e3d2: vTaskEnterCritical at /Users/brennana/amazon-freertos/lib/FreeRTOS/tasks.c:4944

0x40081b76: esp_intr_disable at /Users/brennana/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/intr_alloc.c:776

0x4008be82: spi_intr at /Users/brennana/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/driver/spi_master.c:432

0x40083682: _xt_lowint1 at /Users/brennana/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x4008e415: vTaskExitCritical at /Users/brennana/amazon-freertos/lib/FreeRTOS/tasks.c:4944
```

**Code to reproduce the bug**
This issue should be easy to see from the backtrace and the SPI code block I referenced, but this is function that is sending the request:

```c
void disp_spi_send(uint8_t * data, uint16_t length)
{
	if (length == 0) return;           //no need to send anything

	spi_transaction_t t;
    memset(&t, 0, sizeof(t));       	//Zero out the transaction
	t.length = length * 8;              //Length is in bytes, transaction length is in bits.
	t.tx_buffer = data;               	//Data

	spi_device_queue_trans(spi, &t, portMAX_DELAY);

	spi_transaction_t * rt;
	spi_device_get_trans_result(spi, &rt, portMAX_DELAY);
}
```

rodney-onica, 2019-01-31:
That's exactly what I've been seeing this afternoon since I pulled the most recent code!  

Is it this commit? https://github.com/aws/amazon-freertos/commit/9697cea50e48ceb4ad6c79c05f0446fef857cb25

rodney-onica, 2019-01-31:
@btashton I found this https://github.com/aws/amazon-freertos/issues/62

*The power supply??*   (Whelp.)

Sure enough, I pulled out the USB cable, re-seated it... _problem solved_.

?!?!?!?

Note that it wasn't enough to unplug the cable from the ESP32 itself.

btashton, 2019-01-31:
The root cause of your issue is the same.  The version of FreeRTOS bundled with esp-idf has patches to allow calling `vTaskEnterCritical` from and ISR. 

See this from `task.c`
```cpp
			/* Original FreeRTOS comment, saved for reference:
			This is not the interrupt safe version of the enter critical
			function so	assert() if it is being called from an interrupt
			context.  Only API functions that end in "FromISR" can be used in an
			interrupt. Only assert if the critical nesting count is 1 to
			protect against recursive calls if the assert function also uses a
			critical section. */

			/* DISABLED in the esp32 port - because of SMP, For ESP32
			FreeRTOS, vTaskEnterCritical implements both
			portENTER_CRITICAL and portENTER_CRITICAL_ISR. vTaskEnterCritical
			has to be used in way more places than before, and some are called
			both from ISR as well as non-ISR code, thus we re-organized
			vTaskEnterCritical to also work in ISRs. */
#if 0
			if( newNesting	== 1 )
			{
				portASSERT_IF_IN_ISR();
			}
#endif
```
the spi, rtc, and other isr in the version of esp-idf that amazon-freertos is using are implemented with the expectation that `vTaskEnterCritical` can be called.

Since the Amazon version of FreeRTOS does not implement SMP, some of these calls need to be removed, or `vTaskEnterCritical` needs to support being called from the interrupt context.

qiutongs, 2019-03-20:
close this issue because it is fixed by https://github.com/aws/amazon-freertos/pull/403

closed 2019-03-20 by qiutongs