aws/amazon-freertos #275: TLS send erroneous hard error

mbruno-xmos, 2019-02-04:
During a TLS send if there isn't a free send buffer and the socket is non-blocking an erroneous hard error is triggered and the TLS context is freed. This is the case when publishing messages with the MQTT agent.

See this line:
https://github.com/aws/amazon-freertos/blob/29b654cfa76ad4eca2d11e186f19118c028f0cef/lib/tls/aws_tls.c#L804

mbedtls_ssl_write() does not return 0 when no bytes are sent, but rather -pdFREERTOS_ERRNO_ENOSPC. I have modified the code as follows and this seems to have fixed the problem:
```
            else if( 0 == xResult || -pdFREERTOS_ERRNO_ENOSPC == xResult )
            {
                /* No data sent (and no error). The secure sockets
                 * API supports non-blocking send, so stop the loop but don't
                 * flag an error. */
                xResult = 0;
                break;
            }
```

mradulan, 2019-02-07:
Hi @mbruno-xmos , did you encounter this while running the PubSub demo or a custom application

mbruno-xmos, 2019-02-07:
A custom application. We are publishing larger (~4kbyte) messages multiple times per second.

mbruno-xmos, 2019-02-07:
I have also modified aws_mqtt_agent.c to set the socket to block on sends with a timeout of mqttconfigTCP_SEND_TIMEOUT_MS so that it does not sit in a tight loop in prvMQTTSendCallback(). I'm not sure why it isn't already done that way.

mradulan, 2019-02-07:
Hi @mbruno-xmos , we will be adding the fix for the condition when there is no space available. About adding the timeout in prvMQTTSendCallback, we will have to go over that condition internally again. Thank you for the fix!

mradulan, 2019-03-04:
The issue has been fixed. Please reopen the ticket if the problem persists.

closed 2019-03-04 by mradulan