aws/amazon-freertos #278: OTA out of sync with jobs and not unsubscribing from all topics on failures

nmckendree, 2019-02-06:
**Describe the bug**
OTA library entered state where it was out of sync with the OTA service. The device would immediately reject a new job as it believed it was associated to a previous job that didn't exist or wasn't active any longer. 

> 179776 347670946 [OTA Task] [prvParseJobDoc] Busy with existing job. Ignoring.                  
179777 347670946 [OTA Task] [prvParseJobDoc] Error 1 parsing job document.
179778 347670946 [OTA Task] [prvParseJobDoc] Rejecting job due to OTA_JobParseErr_t 1           
179796 347671098 [OTA Task] [prvOTA_Close] Context->0x005599a4

Additionally, after trying several times i received a subscription manager full error from the mqtt library. 

> 179951 347887000 [OTA Task] [prvParseJobDoc] Job was accepted. Attempting to start transfer.
179952 347887000 [OTA Task] [OTA]	cnewImageStatus = 0
179953 347887092 [OTA Task] [prvPAL_GetPlatformImageState]	currentImageStatus = 0
179954 347887092 [OTA Task] Sending command to MQTT task.
179955 347887092 [MQTT] Received message 16fa0000from queue.
179956 347887092 [MQTT] Initiating MQTT subscribe.
179957 347887093 [MQTT] WARN: Subscription Manager full! No space left to store new subscriptions.
179958 347887093 [MQTT] MQTT_Subscribe failed!

Issues:
Seems like there are no timeouts for the OTA transfer and it is possible to enter some state where the subscriptions are not completely purged from the mqtt agent. Seems that the datastream subscription string is the most likely candidate as this string is unique per job/file chunk. 

**System information**
Amazon FreeRTOS OTA Agent V1.0.0
Amazon FreeRTOS release 1.4.6
Custom platform - Cypress CYW43907

**To reproduce**
1. So far I am unable to develop a chain of events to reproduce this issue. I encountered this problem on two devices out of a pool of 100 performing updates.

mradulan, 2019-02-06:
Hi mckendree, I will add the other two issues #279 and #280 here, as they both stem out of the issue of OTA library being out of sync. Closing the other two issues.

nmckendree, 2019-02-06:
@mradulan, sure no problem. I was trying to keep the issue separated from feature request.

nmckendree, 2019-02-06:
Ran into this again. 
The CYW43907 platform and low level WICED libraries always expect the first block to be received is a header in the first 4k of the image. Due to network traffic or what ever, on some devices this will tend to arrive at a later time resulting in an ota failure. I understand the OTA library doesn't care what order blocks are received in and I am fine with that... non issue. 

After attempting ota about 6 times i ran into the subscription manager error again, this time i printed the subscriptions that were already allocated to the manager. 

1. 1549481350961,$aws/things/CC_C0_79_DF_56_D2/jobs/$next/get/accepted
2. 1549481350961,$aws/things/CC_C0_79_DF_56_D2/jobs/notify-next
3. 1549481350961,$aws/things/CC_C0_79_DF_56_D2/streams/AFR_OTA-cbc84b74-81db-482e-91c9-bef3bc1485fa/
4. 1549481350962,$aws/things/CC_C0_79_DF_56_D2/streams/AFR_OTA-e3722a3a-dd8b-492a-82d5-25eb07165677/
5. 1549481350962,$aws/things/CC_C0_79_DF_56_D2/streams/AFR_OTA-792e2603-8fd8-4314-b430-9a56a2427ca8/
6. 1549481350962,$aws/things/CC_C0_79_DF_56_D2/streams/AFR_OTA-6ca38016-142f-4e67-8dbc-c5858f4c4be9/
7. 1549481350962,$aws/things/CC_C0_79_DF_56_D2/streams/AFR_OTA-79270fea-cd35-4ff8-8d82-f622ad945151/
8. 1549481350962,$aws/things/CC_C0_79_DF_56_D2/streams/AFR_OTA-f11e2369-d76d-4752-9fea-2f152ffbd5b5/

This clearly points to stream subscriptions being stuck in the manager with no way to remove them unless the unit either drops the mqtt connection (guessing this will clear the manager but not sure), or reset the board. 

Thoughts?

mradulan, 2019-02-06:
yes it looks like the mqtt subscription manager should remove the jobs which are not active currently. We are debugging the code to why it is behaving incorrectly.

nmckendree, 2019-02-07:
After looking at the chain of events in my serial log, I can see when there are error conditions in the OTA library, the Unsubscribe feature may fail. There are no retries in the OTA library so the topic becomes stuck in the subscription manager.

nmckendree, 2019-02-07:
In aws_ota_agent.c, 

I added retry loop to all subscribe, publish, and unsubscribe attempts (max 5 attempts), with a random 1s - 2s delay between attempts. This seems to resolve the issue. 
It looks like the biggest issue relates to the failure to the publish the "FAILURE" status message, so the server keeps pumping blocks of data out. By adding the publish retry, the server eventually processes the publish message and responds with a PUBACK and then the unsubscribe works. 

I can replicate this by modifying my prvPAL_WriteBlock to fail after some number of bytes written, like 12K while sending a file a order of magnitude larger, like 512K. 

[logCapture.txt](https://github.com/aws/amazon-freertos/files/2842702/logCapture.txt)

WRansohoff, 2019-02-13:
Thank you for the detailed report on how to reproduce this issue! We will look more closely at what happens when the prvPAL_WriteBlock method fails partway through an update as we continue to investigate.

qiutongs, 2019-03-20:
We have figured out the root cause and are working on the fix. But no ETA so far.

pvyawaha, 2019-12-24:
Hi nmckendree,

Do you still have observe this issue when doing OTA ? Which Amazon FreeRTOS release are you currently using ?

qiutongs, 2020-02-08:
Thanks for reporting the bug. We are still actively working on this and will merge the fix to master in coming releases.

closed 2020-02-08 by qiutongs