aws/amazon-freertos #297: Nordic NRF52 DK support

noblerabbit, 2019-02-11:
Hi,

I pulled the feature/ble-beta branch for the mqtt over ble example.

I got the vStartMQTTBLEEchoDemo(); working on ESP32. I have a functional ios aws demo app, and aws credentials in order so I can listen and post to freertos/demos/echo.

Now I am trying to test also the Nordic52 DK. 

Following the aws instructions again and using SEGGER Studio on my Mac (open aws_demos_ble.emProject) I am able to build and run the 'aws_demos_ble' on the board but I don't get any output on the terminal (even if I press reset button) and the device is not broadcasting.

I tested 'ble_app_hrs' from nordic SDK and it is working well (broadcasting and I can also see the output on the terminal).

Anyone successfully tested nordic52 dk? Thanks.

noblerabbit, 2019-02-11:
Here I also a screen shoot when I do Debug>GO. Thanks.
[image]

WRansohoff, 2019-02-12:
Hi there, thanks for reaching out - I have started setting up an environment to try to reproduce the issue that you are experiencing.

In the meantime, would you mind providing a little more information about the build output that you see? Some of our demos set an automatic breakpoint near the beginning of their 'main' methods, so could you double-check whether anything changes if you click on the green arrow in the 'Disassembly' panel's title bar, circled in purple below?

[image]

Thank you for your patience while we investigate.

noblerabbit, 2019-02-13:
Hi,

Thanks for the fast response.

I pulled the latest feature/ble-beta this morning. It again stops at 0x978. If I press "play" it is "Running..." but I don't see it broadcasting and there is no output on the terminal (even if i am pressing the reset button on the nrf52).

Thanks.

[image]

WRansohoff, 2019-02-13:
Great, thank you for the information and patience. I was able to build this demo for the Nordic dev kit, and I did see a similar issue when I tried to run it on a new board.

It looks like there is an additional step which we need to perform before using the Amazon FreeRTOS demos with the Nordic devkit; we have a 'bootloader' project which needs to be built and flashed to the board before the demo applications. If you are using Segger Embedded Studio, the bootloader project file should be located at:

demos/nordic/nrf52840-dk/bootloader/pca10056/ses/bootloader_mbr_pca10056.emproject

You should be able to build it in the same way as the demo project, and it looks like clicking 'Debug -> Go' should be enough to flash the bootloader to the board once it builds.

So does the project work for you if you build and flash the bootloader project, then re-build and re-flash the demo application? Thanks again for bearing with us while this feature is in beta.

noblerabbit, 2019-02-13:
@WRansohoff I followed your instructions and flashed the nrf52 first with bootloader and then with the "aws_demos_ble". Indeed the device is broadcasting and I can also see the output on the terminal.

Now I get I get "UNKNOWN BLE EVENT 35" when I connect to the device the FreeRTOS ios demo app. I cannot toggle "Enable MQTT Proxy" in the app.

Attached the screenshots. Thanks.
[image]

[image]

WRansohoff, 2019-02-14:
Well, that's progress! Sorry to hear that you're still having trouble, though - could you provide a little more information about the iOS device that you are using?

The NRF52840 devkit demo currently requires that the host device supports Bluetooth version 4.2 or later, and it looks like that support varies depending on what type of iOS device you are using. Would you mind telling us what sort of iOS device you are running the SDK app on, and what version of iOS it is running on? Thanks again for your patience.

noblerabbit, 2019-02-14:
@WRansohoff sure. I am using iPhone 7, iOS 12.1.4.

WRansohoff, 2019-02-14:
Okay, it does look like that device should support BLE v4.2. Most mobile devices like the iPhone do have a 'Bluetooth cache' which can sometimes store stale or buggy data and cause issues. So, would you mind rebooting the iPhone to clear that cache?

If that doesn't help, could you try using a Bluetooth diagnostics app to display the characteristics which are being sent by the Nordic devkit? That might give us more information to work with than the demo SDK app's output. Thanks.

noblerabbit, 2019-02-14:
@WRansohoff I went to Bluetooth settings in the iPhone, looked for NRF52 and clicked "Forget this device", then restarted the iPhone and it works now! Thanks! 👍

FYI another bug I found that is not directly related to this ticket: I noticed that "python SetupAWS setup" does not generate correct clientcredentialMQTT_BROKER_ENDPOINT[]. I had to insert the "-ats" part manually (I compared the endpoint on the IoT Core, with the one generated by the script.)

closed 2019-02-15 by noblerabbit

WRansohoff, 2019-02-15:
Great, I'm glad to hear that you got it working! And thank you for reporting the setup script bug - we are tracking that in issue #299 and our current plan is to have the script generate '-ats' endpoints by default.