aws/amazon-freertos #299: SetupAWS.py setup not using the -ats endpoints

teuteuguy, 2019-02-12:
AWS moved the endpoints to -ats because of the web consortium move off of Verisign root CAs.

The SetupAWS.py script gets and still writes the old endpoint (without the -ats) into the header files.

Whilst generating certificates off of the new root CAs.

mradulan, 2019-02-12:
thank you for pointing this. Although ATS is now available, Verisign also remains in regions launched before May 2018. We probably need to provide options for ATS and verisign.

teuteuguy, 2019-02-12:
Agreed, however simply running the script and expecting the code to run, will not work, because of mismatch between endpoint and generated certificates.

For now, simple fix, after running the script is to manually modify the aws_clientcredential.h file and add the "-ats" after the custom portion of the endpoint.

Best, simply modify the script to get the right endpoint and write it automatically to the aws_clientcredential.h file.

mradulan, 2019-02-12:
The device chooses which CAs to trust according to the following rules:
1. On the platforms using mbedTLS, we trust both Verisign and ATS. Depending on which endpoint you connect to, the server will present either Verisign or ATS signed certificate and the device will trust it.
2. On ST and TI (platforms where we offload SSL), we decide which CA to trust based on the endpoint you are using - if you are using ATS endpoint (endpoint containing "-ats.iot"), we trust ATS otherwise we trust Verisign. In this case, the expectation is:
  - While connecting to non-ATS endpoint – The server presents Verisign signed certificate.
  - While connecting to ATS endpoint – The server presents ATS signed certificate.

This, however, is not true in the ATS only regions(eg. ap-south-1 https://aws.amazon.com/blogs/iot/aws-iot-core-ats-endpoints/). In these regions, the server will present ATS signed certificate regardless of which endpoint you are using. If you happen to use non ATS endpoint, the device will incorrectly reject the server certificate because the device was expecting a Verisign signed certificate. 

We will update the setup script to use ATS region by default as new regions will support only that.

WRansohoff, 2019-02-15:
Okay, this should be fixed by PR #301 - but please feel free to re-open this if you still see issues. Thank you for reporting this.

closed 2019-02-15 by WRansohoff