aws/amazon-freertos #303: Windows Device Simulator -  MQTT echo failed to connect with error

matriedel, 2019-02-13:
ERROR:  MQTT echo failed to connect with error

-- log from aws_demos.exe ---

The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which
should be defined in FreeRTOSConfig.h
Attempting to open interface number 1.
2 132 [IP-task] vDHCPProcess: offer 192.168.158.50
3 172 [IP-task] vDHCPProcess: offer 192.168.158.50
4 172 [IP-task] Write code signing certificate...
5 172 [IP-task] Key provisioning done...
6 172 [IP-task] Creating MQTT Echo Task...
7 172 [IP-task]

IP Address: 192.168.158.50
8 172 [IP-task] Subnet Mask: 255.255.255.0
9 172 [IP-task] Gateway Address: 192.168.158.1
10 172 [IP-task] DNS Server Address: 192.168.158.1

11 172 [MQTTEcho] MQTT echo attempting to connect to Paste AWS IoT Broker endpoint here..
12 5292 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
13 5292 [MQTTEcho] MQTT echo test could not connect to broker.
14 5292 [MQTTEcho] MQTT echo demo finished.
15 5292 [MQTTEcho] ----Demo finished----

----------------  Log from Interface --------
[image]

Any hint? 

Best

Matthias

matriedel, 2019-02-13:
Setting 
#define mqttconfigENABLE_DEBUG_LOGS    1
"MQTT echo test could not connect to broker" seems to be the cause of the problem

IP Address: 192.168.158.50
8 193 [IP-task] Subnet Mask: 255.255.255.0
9 193 [IP-task] Gateway Address: 192.168.158.1
10 193 [IP-task] DNS Server Address: 192.168.158.1

11 193 [MQTTEcho] MQTT echo attempting to connect to Paste AWS IoT Broker endpoint here..
12 193 [MQTTEcho] Sending command to MQTT task.
13 193 [MQTT] Received message 10000 from queue.
14 333 [IP-task] Socket sending wakeup to MQTT task.
15 5333 [MQTT] About to close socket.
16 5333 [MQTT] Socket closed.
17 5333 [MQTT] Stack high watermark for MQTT task: 238
18 5333 [MQTT] Notifying task.
19 5333 [MQTT] Received message 0 from queue.
20 5333 [MQTTEcho] Command sent to MQTT task failed.
21 5333 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
22 5333 [MQTTEcho] MQTT echo test could not connect to broker.
23 5333 [MQTTEcho] Sending command to MQTT task.
24 5333 [MQTT] Received message 20000 from queue.
25 5333 [MQTT] Notifying task.
26 5333 [MQTTEcho] Command sent to MQTT task passed.
27 5333 [MQTTEcho] MQTT echo demo finished.
28 5333 [MQTTEcho] ----Demo finished----

aggarg, 2019-02-13:
>> 11 172 [MQTTEcho] MQTT echo attempting to connect to Paste AWS IoT Broker endpoint here..

Looking at the above logs, it seems that you have not configured your endpoint. Would you please use our onboarding workflow to setup your AWS IoT resources. The steps for the same are mentioned in the following document: [OnBoardingWorkflow-Win.pdf](https://github.com/aws/amazon-freertos/files/2861461/OnBoardingWorkflow-Win.pdf)

Let us know if that works.

Thanks.

matriedel, 2019-02-14:
I already went along the "quick connect" installation approach. 
Maybe the log from Interface where you find the destination address is 0.0.0.0 is a hint of the issue

matriedel, 2019-02-14:
I cleaned up everything and started again from scratch. Now it worked

closed 2019-02-14 by matriedel