aws/amazon-freertos #304: Improve feedback for Shadow Client Failure

tombresson, 2019-02-13:
Problem:

During implementation of the Shadow Client, the initial push of shadow data would indicate "Timed out". This clearly was not the case in the backend, as the data would actually make it to AWS (as it was seen appear on the Device Shadow). 

The real issue was, JSMN was not allocated enough tokens and was failing to tokenize the response that is received on the `shadow/update/accepted` topic. The tokenization is done in the the MQTT callback, but the JSMN feedback is not passed back to the Shadow Client. 

The semaphore wait times out (see relevant lines linked below) long after the MQTT Client has called back to `prvShadowUpdateCallback` (through a series of callback function calls). The description of the header related to the `SHADOW_Update` function don't give indication of this being a possible failure case. Funny enough, the JSMN errors have been defined in the header (`aws_shadow.h`) further up in the file.

https://github.com/aws/amazon-freertos/blob/c542bafb75b774ec3f4b3e2de17f909169654123/lib/shadow/aws_shadow.c#L1242-L1249

**Describe the solution you would like.**  
The feedback for the failure case could be much more indicative of the specific failure instead of simply "timed out".

WRansohoff, 2019-02-13:
Hi Tom,

Thank you for the feedback and detailed summary - if I'm understanding you right, it sounds like our error reporting logic in the Shadow client is missing the type of error that your application ran into.

Looking at the code snippet you linked to, it seems like the Shadow client reports a 'timeout' when it cannot acquire the relevant semaphore, and I can definitely see that sounding confusing without more details about why the semaphore was never released; it sounds a lot like the client is reporting a network timeout.

Thanks for reporting this, we want our error messages to be as helpful as possible. We'll look into how this message is phrased, and whether we can catch the errors you describe earlier to provide a clearer picture of what went wrong.

tombresson, 2019-02-15:
Hi @WRansohoff,

I think the key takeaway is, in the Shadow Client lib:

* Semaphore is taken by the application call into the Shadow Client
* The message is received on the `/update/accepted` or `/update/rejected` topic
* Calback fails to tokenize the payload received on the topic **because there are too few JSMN tokens allocated in the data array.**
* Callback exits **without giving the semaphore** because the message didn't contain the required `client-token` 
  * What really happened is the data did contain the correct `client-token` but JSMN failed
* Semaphore wait times out for the application call with the feedback of a timeout (instead of what actually happened)

Thanks for your support.

-Tom

WRansohoff, 2019-02-15:
Okay, we can definitely update the error message to make things more clear - I've proposed an update to that wording in pull request #312

To clarify the rest of your points, was there any other debugging information printed out by the application to alert you to the JSMN failure? I could be reading it wrong or we might need more debug logging in places, but it looks like the library is supposed to print debugging information about individual failures down the stack as they happen, like in the 'prvGetErrorCodeAndMessage' method. Does that not happen with the type of failure that you saw, or are there other places where you would like to see more detailed logging?

Thanks for the feedback.

tombresson, 2019-02-15:
No other debug information was printed out when this occurs. 

The relevant code is below:

This is called to do the tokenization. When it fails, it returns `pdFAIL` to...
https://github.com/aws/amazon-freertos/blob/a81bf209db6e86a75729f910443a1dc016fc737d/lib/shadow/aws_shadow_json.c#L72-L124

this function. This function then simply exits silently. I';d guess this is probably by design as other messages received on the topic not matching the message with the intended `client-token` value would trigger a message as well.
https://github.com/aws/amazon-freertos/blob/a81bf209db6e86a75729f910443a1dc016fc737d/lib/shadow/aws_shadow.c#L1042-L1062

WRansohoff, 2019-02-15:
Okay, then it sounds like we need some debug logging around this error case - thanks for catching that! It looks like the `prvParseJSON` method calls `jsmn_parse` and returns that value, so we should be able to add a debug statement around the case where JSON parsing returns `JSMN_ERROR_NOMEM`; that looks like what JSMN should return when it is not allocated enough tokens:

https://github.com/aws/amazon-freertos/blob/a81bf209db6e86a75729f910443a1dc016fc737d/lib/third_party/jsmn/jsmn.h#L50-L57

Does that sound like it would have made your issue easier to debug?

tombresson, 2019-02-15:
If the call to JSMN caught the failure and printed a debug message, yes. That could have make the whole ambiguous "Got a timeout but the shadow was still updated" issue easier to resolve.

Thanks @WRansohoff.

WRansohoff, 2019-02-15:
Sounds good, I updated PR #312 to add a debug logging statement which reports the JSMN error code if the `jsmn_parse` method fails. Thanks for the feedback, and please let us know if you encounter any more issues with this.

WRansohoff, 2019-02-19:
Closing this issue because PR #312 has been merged in, but please let us know if you run into any further issues with this.

closed 2019-02-19 by WRansohoff