aws/amazon-freertos #317: ESP32 DevKitC - [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.

borch84, 2019-02-16:
I am running the steps in order to connect to MQTT broker. I am following this steps:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

When esp32 is trying to connect to MQTT broker it is showing [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.

**System information**
- ESP32 DevKitC V1
- Operating System Linux
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it): v1.4.6-58-ga81bf209
- Project MQTT Sub pub Demo

**Expected behavior**
Demo should connect to MQTT broker

**Screenshots or console output**
I (8534) WIFI: SYSTEM_EVENT_STA_GOT_IP
2 840 [IP-task] vDHCPProcess: offer c0a8046aip
3 841 [main] WiFi Connected to AP. Creating tasks which use network...
4 842 [main] Write certificate...
I (8554) PKCS11: Initializing NVS partition: "storage"
5 846 [main] Write device private key...
6 856 [main] Creating MQTT Echo Task...
7 856 [MQTTEcho] MQTT echo attempting to connect to XXXXXXXXXX-ats.iot.us-east-1.amazonaws.com.
8 856 [MQTTEcho] Sending command to MQTT task.
9 857 [MQTT] Received message 10000 from queue.
10 1858 [IP-task] Socket sending wakeup to MQTT task.
11 2358 [MQTT] About to close socket.
12 2358 [MQTT] Socket closed.
13 2358 [MQTT] Stack high watermark for MQTT task: 4696
14 2358 [MQTT] Notifying task.
15 2358 [MQTT] Received message 0 from queue.
16 2358 [MQTTEcho] Command sent to MQTT task failed.
17 2358 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
18 2358 [MQTTEcho] MQTT echo test could not connect to broker.
19 2358 [MQTTEcho] Sending command to MQTT task.
20 2359 [MQTT] Received message 20000 from queue.
21 2359 [MQTT] Notifying task.
22 2360 [MQTTEcho] Command sent to MQTT task passed.
23 2360 [MQTTEcho] MQTT echo demo finished.
24 2360 [MQTTEcho] ----Demo finished----

I am testing the connection from my computer using mosquitto_pub client and the connection is successful:

$ mosquitto_pub -d --cafile AmazonRootCA1.pem --cert esp32-reedswitch-aws-certificate/27eb999b86-certificate.pem.crt --key esp32-reedswitch-aws-certificate/27eb999b86-private.pem.key -h XXXXXXXX-ats.iot.us-east-1.amazonaws.com -p 8883 -m "test message" -t "freertos/demo/echo" -q 1

Client mosqpub/14670-g4560 sending CONNECT
Client mosqpub/14670-g4560 received CONNACK
Client mosqpub/14670-g4560 sending PUBLISH (d0, q1, r0, m1, 'freertos/demo/echo', ... (12 bytes))
Client mosqpub/14670-g4560 received PUBACK (Mid: 1)
Client mosqpub/14670-g4560 sending DISCONNECT

Thank you!

WRansohoff, 2019-02-18:
Hi, thanks for reaching out - I'm sorry to hear that you're having issues.

MQTT broker connection problems are often related to device credentials, so would you mind double-checking a few values in your current setup? If you ran `SetupAWS.py cleanup` to clear the credentials out of your configuration files, some of these values might be empty until you run `SetupAWS.py setup` again.

* In the `demos/common/include/aws_clientcredential.h` file, do the MQTT endpoint, Thing name, and WiFi credential values all look correct?

* In `demos/common/include/aws_clientcredential_keys.h`, do you see the certificate and RSA private key which were configured for your device and which worked with your mosquito client connection?

* Can you verify that the setup script created a 'Thing' record in your AWS 'IoT Core' dashboard? (Click 'Manage' on the left sidebar to view all of your Things.)

Thank you for your patience, and feel free to let us know if any of those questions are unclear.

borch84, 2019-02-19:
hi @WRansohoff, this is the content of aws_clientcredential.k:

#ifndef __AWS_CLIENTCREDENTIAL__H__
#define __AWS_CLIENTCREDENTIAL__H__
#include "aws_clientcredential_keys.h"
#define clientcredentialIOT_THING_NAME "mythingname"
#define clientcredentialMQTT_BROKER_PORT 8883
#define clientcredentialGREENGRASS_DISCOVERY_PORT 8443
#define clientcredentialWIFI_SSID       "myssid"
#define clientcredentialWIFI_PASSWORD   "myssidpassword"
#define clientcredentialWIFI_SECURITY   eWiFiSecurityWPA
#endif

my aws_clientcredential_key.h is:

/*

#ifndef AWS_CLIENT_CREDENTIAL_KEYS_H
#define AWS_CLIENT_CREDENTIAL_KEYS_H

#include <stdint.h>

/*
#define keyCLIENT_CERTIFICATE_PEM \
"-----BEGIN CERTIFICATE-----\n"\
"MIIDWjCCAkKgAwIBAgIVAMZGI52JduxC6yIvir4ccTPoJIkmMA0GCSqGSIb3DQEB\n"\
...
"j0AZLKTdyH+YUBxY+H0/5ohH4LQtbjbZ0wicnl8wvxih1DO+rW2RQhNHME4ezA==\n"\
"-----END CERTIFICATE-----\n"

/*
#define keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM  NULL

/*
#define keyCLIENT_PRIVATE_KEY_PEM \
"-----BEGIN RSA PRIVATE KEY-----\n"\
"MIIEowIBAAKCAQEAo6T61BDyXjjCL5Afj/dCEQlz5q0hPtm5wwCkrO4gbtCbRLZj\n"\
...
"ur4oKUxXiEH15GAPo9HIenR3i256FSq7Xz3ufDRgScEdKb50oLFo\n"\
"-----END RSA PRIVATE KEY-----\n"

/* The constants above are set to const char * pointers defined in aws_demo_runner.c,
extern const char clientcredentialCLIENT_CERTIFICATE_PEM[];
extern const char* clientcredentialJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM;
extern const char clientcredentialCLIENT_PRIVATE_KEY_PEM[];
extern const uint32_t clientcredentialCLIENT_CERTIFICATE_LENGTH;
extern const uint32_t clientcredentialCLIENT_PRIVATE_KEY_LENGTH;

#endif /* AWS_CLIENT_CREDENTIAL_KEYS_H */
`

The setup script created the thing, certificate and policies. I can see it from my IoT Core Console. 

Thanks

WRansohoff, 2019-02-19:
Okay, that does look correct - and just to make sure, in the `aws_clientcredential.h` file, there should also be a line which looks like this, with the correct AWS region:

`static const char clientcredentialMQTT_BROKER_ENDPOINT[] = "xxxx-ats.iot.[region].amazonaws.com";`

If your mosquito client is still able to publish a message successfully, then it sounds like the AWS permissions are probably set up correctly with your endpoint and certificates.

So just to double-check, could you also verify that the `IDF_PATH` environment variable is not set when you build the ESP32 demo project? There is a specific version of ESP-IDF included in the Amazon FreeRTOS repository under `lib/third_party/mcu_vendor/espressif/esp-idf`, and sometimes issues can emerge if a different version of the toolchain is already installed and being used.

Thanks for your patience.

borch84, 2019-02-19:
hi @WRansohoff 

aws_clientcredential.h is:

grep MQTT_BROKER aws_clientcredential.h 
static const char clientcredentialMQTT_BROKER_ENDPOINT[] = "XXXXXXX-ats.iot.us-east-1.amazonaws.com";

Before compiling amazon-freertos I unset the variable with: unset IDF_PATH
Compilation shows warnings,  but my ESP32 get flashed with no issues. ESP32 connects successfully to my wifi and it fails when connecting to mqtt broker.

WRansohoff, 2019-02-20:
Huh, that does all look like it should work - which version of the Amazon FreeRTOS repository are you using? (You can check the commit hash that you're on with `git rev-parse HEAD`.)

borch84, 2019-02-20:
Hi @WRansohoff, it is: git rev-parse HEAD
f0c0efc79bcd2f506186e47b6f2989eb58b25ed6

sarenameas, 2019-03-18:
Hi, are you still having issues? Could you let me know what the policy attached to your thing's certificate is?
To use the echo demo we have users configure the policy from https://docs.aws.amazon.com/freertos/latest/userguide/get-started-freertos-thing.html:
```
{
    "Version": "2012-10-17",
    "Statement": [
    {
        "Effect": "Allow",
        "Action": "iot:Connect",
        "Resource":"arn:aws:iot:<aws-region>:<aws-account-id>:*"
	}, 
    {
        "Effect": "Allow",
        "Action": "iot:Publish",
        "Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:*"
    },
    {
         "Effect": "Allow",
         "Action": "iot:Subscribe",
         "Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:*"
    },
    {
         "Effect": "Allow",
         "Action": "iot:Receive",
         "Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:*"
    }
    ]
}
```
Please make sure this is correct. We typically have connection errors when this is the case. 

Although it is not recommended, you can also allow everything just to narrow down that your policy is indeed the issue:
```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": [
        "iot:*"
      ],
      "Resource": "*"
    }
  ]
}
```

closed 2019-04-29 by dcgaws