aws/amazon-freertos #322: AWS_Shadow

egoltzman, 2019-02-19:
I based my device code on the shadow demo, I saw that when I'm losing connection (MQTT disconnected) I can not subscribe to any callback and thus can not know that this is the case.
I browse the AWS_Shadow.c code and found this (starting from line 883 V1.0.5):
```
    /* The Shadow Client assumes all subscriptions are lost on disconnect. */
    else
    {
        if( pxCallbackParams->xMQTTEvent == eMQTTAgentDisconnect )
        {
            Shadow_debug_printf( ( "[Shadow %d] Warning: got an MQTT disconnect"
                                   " message.\r\n", xShadowClientID ) );
            prvSetSubscribedFlag( pxShadowClient, eShadowOperationUpdate, 0 );
            prvSetSubscribedFlag( pxShadowClient, eShadowOperationGet, 0 );
            prvSetSubscribedFlag( pxShadowClient, eShadowOperationDelete, 0 );
              }

            /*_RB_ TODO below. */
            /* TODO: resubscribe to all callback topics. */
        }
```
I would expect that the module will call a callback (from this location) to signal that situation but I can't see one, I can see that there is a TODO here regarding resubscribe which is also an issue after MQTT is disconnected. 

Is there any other mechanism that can help in this situation or should the AWS_Shadow.c need to be updated with something to handle it?

aggarg, 2019-02-20:
Please see this answer here: https://github.com/aws/amazon-freertos/issues/326#issuecomment-465749771

Please let us know if you have any feedback about the new library. We are very interested in your feedback.

Thanks.

sarenameas, 2019-03-16:
Hello have you had a chance to look at the latest implementation of the Shadow library addressed in https://github.com/aws/amazon-freertos/issues/326#issuecomment-465749771 ? I will be closing this issue in the meantime.

closed 2019-03-16 by sarenameas