aws/amazon-freertos #323: MQTT echo connected but could not subscribe to freertos/demos/echo

temanXMC, 2019-02-20:
Hello,

i am using Infineon XMC4800 IoT connectivity Kit for connecting AWS cloud. i am following instructions as shown in Amazon FreeRTOS user guide and so encounter an error (MQTT echo demo could not subscribe to freertos/demos/echo)

have i missed steps in setting up connection ? could you advise ? 
Thank you in advance.

[error.pdf](https://github.com/aws/amazon-freertos/files/2882674/error.pdf)

Best Regards,
Felix

aggarg, 2019-02-20:
Hi,

One possible reason for not being able to subscribe may be that the policy is not correctly setup. Would you please use our onboarding workflow to ensure that all the AWS IoT resources and the corresponding credentials are setup correctly. The steps for the same are mentioned in the following document: [OnBoardingWorkflow-ST.pdf](https://github.com/aws/amazon-freertos/files/2883310/OnBoardingWorkflow-ST.pdf)

Please note that the document mentions “ST” for the hardware platform while you should choose “Infineon” instead.

Thanks.

temanXMC, 2019-02-21:
hi Aggarg,

Thank you for comments. Right, i included the following in policy and it works.

"Version": "2012-10-17",
"Statement": [
{
"Effect": "Allow",
"Action": "iot:Connect",
"Resource":"arn:aws:iot:<aws-region>:<aws-account-id>:*"
},
{
"Effect": "Allow",
"Action": "iot:Publish",
"Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:*"
},
{
"Effect": "Allow",
"Action": "iot:Subscribe",
"Resource": "arn:aws:iot:<aws-region>>:<aws-account-id>:*"
},
{
"Effect": "Allow",
"Action": "iot:Receive",
"Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:*"
}
]

Felix

aggarg, 2019-02-21:
I am glad that it worked for you. 

Thanks.

closed 2019-02-21 by aggarg