aws/amazon-freertos #326: Feedback from Shadow Client for MQTT Disconnect

tombresson, 2019-02-20:
### Desired Feature Description

Because of the Device Shadow client consuming the MQTT callback, no feedback is provided to the application level if an MQTT disconnect occurs. The callback functionality only reaches back as far as `prvShadowMQTTCallback`. This function could/should report the disconnect back to the application so it can reconnect, if desired.

## Describe the solution you would like

Shadow Client could contain reference to an application call-back to handoff the MQTT callback to if there is nothing of interest for the Shadow Client in the callback.

This would allow the application to respond to `eMQTTAgentDisconnect` events in an alternative way other than [simply printing `"Warning: got an MQTT disconnect message."`](https://github.com/aws/amazon-freertos/blob/f0c0efc79bcd2f506186e47b6f2989eb58b25ed6/lib/shadow/aws_shadow.c#L882)

## Additional context

(I believe) Currently without a way to detect an MQTT disconnect, the application can not detect the disconnect without first failing to publish. If this is incorrect, I do not see another way the application can detect this failure.

lt72, 2019-02-20:
@tombresson: We do have a new version of MQTT and Shadow library which provide a finer level of control as far as the user callback is concerned. It exists in BLE-Beta branch and is currently work in progress: https://github.com/aws/amazon-freertos/blob/feature/ble-beta/lib/include/aws_iot_mqtt.h 
I would suggest you adopting the new library, and please let us know of any positive/negative feedback.
Documentation is here: https://docs.aws.amazon.com/freertos/latest/userguide/freertos-mqtt-2.html 
API reference is here: https://docs.aws.amazon.com/freertos/latest/lib-ref/html3/mqtt/index.html 
We are very interested in your feedback.

tombresson, 2019-02-20:
By the way, I apologize, but this issue/feature request appears to already be mentioned in #322.

sarenameas, 2019-03-16:
Hello, have you had a change to look at the latest implementation of the shadow library mention by lt72? I will be closing this issue in the meantime.

closed 2019-03-16 by sarenameas