aws/amazon-freertos #327: Shared MQTT Client with Shadow Client

tombresson, 2019-02-20:
## Detailed feature description
A clear and concise description of what the problem is. What are the use cases of the feature to be added? 

There is a need to share the MQTT Agent that is created to service the Device Shadow (made through the Shadow Client lib) with another functionality in the system.

It appears [this functionality was thought about in the case of the Shadow Client lib](https://github.com/aws/amazon-freertos/blob/f0c0efc79bcd2f506186e47b6f2989eb58b25ed6/lib/include/aws_shadow.h#L107), but has not been implemented. 

This functionality is necessary, as both the Shadow functionality and the additional MQTT Topic operate on the AWS at the same endpoint/port. Operating them both independently through the Shadow Client and the MQTT Agent causes them to step on each other, as connect and disconnects occur. 

## Describe the solution you would like

A clear and concise method, either at the Shadow Client or the MQTT Agent level, to connect to the same end point and service MQTT connections simultaneously. 

As it stands right now, the "sharing" will have to be implemented in the application, and it will only be a minimally capable sharing. Currently [the application doesn't have access to the MQTT callback](https://github.com/aws/amazon-freertos/issues/326) so disconnects can't be easily recognized, which makes managing both of these connection more difficult. 

**Additional context**  
Thank you! 😄

lt72, 2019-02-20:
@tombresson: AS I mentioned for issue #326, we do have a new version of MQTT and Shadow library which provide a finer level of control as far as the user callback is concerned. It exists in BLE-Beta branch and is currently work in progress: https://github.com/aws/amazon-freertos/blob/feature/ble-beta/lib/include/aws_iot_mqtt.h 
The new library allows sharing connections. 
I would suggest you adopting the new library, and please let us know of any positive/negative feedback.
Documentation is here: https://docs.aws.amazon.com/freertos/latest/userguide/freertos-mqtt-2.html
API reference is here: https://docs.aws.amazon.com/freertos/latest/lib-ref/html3/mqtt/index.html
We are very interested in your feedback. 

Please note that creating the connection is currently a step-by-step process, but we are working on adding a factory method that will make that more user friendly, i.e. calling one API only.

tombresson, 2019-02-20:
Hi @lt72. Thanks for responding (to both feature requests). I'll certainly consider adopting the new library. I was taking a look at it today, just to see what has changed.  

Ultimately, will the API remain the same for either the Shadow Client lib and/or the MQTT Agent lib, or will it require a re-implementation of the use of the libraries? 

I'm not willing to provide feedback on the beta library, but if it saves me effort in waiting for the library to make it into a release build, I would be more likely to take that path. This is simply based on effort allowed for rework and refactoring.

lt72, 2019-02-21:
Hi @tombresson: waiting for the library to get into release will definitely save you time, so I suggest you wait. You can take a look at the header, to gauge the work you'll need to do and, most important, get a feel for whether your needs are met or not.
Header fo rnew MQTT lib is here: https://github.com/aws/amazon-freertos/blob/feature/ble-beta/lib/include/aws_iot_mqtt.h 
Here is Shadow: https://github.com/aws/amazon-freertos/blob/feature/ble-beta/lib/include/aws_iot_shadow.h 
Signature may mildly vary before we declare this a release candidate. 

The current API still exists, btw, and it is implemented in top of the new one for backward compatibility.

lt72, 2019-03-07:
Hi @tombresson, I was wondering if you had a chance to review the libs and if you have any questions. I will close the ticket if I do not hear from you in a couple of days, but please feel free to re-open if you need to.

tombresson, 2019-03-07:
Hey @lt72, I had been in touch with @dnoal regarding our transition timeline. You can close this ticket, as it sounds like it will be addressed in the MQTT lib changes coming soon. Thanks!

-Tom

closed 2019-03-08 by gordonwang0

egoltzman, 2019-05-28:
I'm not sure if I have the same issue as Tom or a different one but it is look like the same so I want to ask this here:
I have a shadow connection that I use in my application, in parallel to that connection I need to publish some ad hoc topics for handling special application issues (non-shadow custom topics).
When I used my initial end point URL I was able to create both a shadow client as well as an MQTT one and implement this need.  Lately I created some more things (since then I used a single thing that was created a long time ago and was used for the development) and for them I was forced to use my end point with the "-ats" extension, but now, from some reason the shadow connection fail if the MQTT one is connected.
My first question is why that and if there is a walk around it.
My second question is, based on the answer to Tom, will I be able to use the shadow MQTT connection for those custom topics in the next version?
Thanks,
Eyal