aws/amazon-freertos #338: Building ESP32 demo failed

Benji1992-HU, 2019-02-22:
Hy!
I tried to make and flash the ESP32 demo project. The console give me some error:
# make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/Benji/esp/esp-idf/requirements.txt are satisfied.
Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "aws_demos" version: 1
CC build/esp_event/default_event_loop.o
AR build/esp_event/libesp_event.a
CC build/esp_http_client/esp_http_client.o
CC build/esp_http_client/lib/http_utils.o
CC build/esp_http_client/lib/http_auth.o
In file included from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:43:0,
                 from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:46,
                 from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:21:
C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip4_addr.h:51:8: error: redefinition of 'struct ip4_addr'
 struct ip4_addr {
        ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:20:0:
C:/amazon-freertos-master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/tcpip_adapter/include/tcpip_adapter.h:51:16: note: originally defined here
 typedef struct ip4_addr {
                ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:43:0,
                 from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:46,
                 from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:21:
C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip4_addr.h:57:25: error: conflicting types for 'ip4_addr_t'
 typedef struct ip4_addr ip4_addr_t;
                         ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:20:0:
C:/amazon-freertos-master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/tcpip_adapter/include/tcpip_adapter.h:53:3: note: previous declaration of 'ip4_addr_t' was here
 } ip4_addr_t;
   ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:44:0,
                 from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:46,
                 from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:21:
C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip6_addr.h:58:8: error: redefinition of 'struct ip6_addr'
 struct ip6_addr {
        ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:20:0:
C:/amazon-freertos-master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/tcpip_adapter/include/tcpip_adapter.h:55:16: note: originally defined here
 typedef struct ip6_addr {
                ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:44:0,
                 from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:46,
                 from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:21:
C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip6_addr.h:63:25: error: conflicting types for 'ip6_addr_t'
 typedef struct ip6_addr ip6_addr_t;
                         ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:20:0:
C:/amazon-freertos-master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/tcpip_adapter/include/tcpip_adapter.h:57:3: note: previous declaration of 'ip6_addr_t' was here
 } ip6_addr_t;
   ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:44:0,
                 from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:46,
                 from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:21:
C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip6_addr.h:78:0: warning: "IP6_ADDR_BLOCK1" redefined
 #define IP6_ADDR_BLOCK1(ip6addr) ((u16_t)((lwip_htonl((ip6addr)->addr[0]) >> 16) & 0xffff))
 ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:20:0:
C:/amazon-freertos-master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/tcpip_adapter/include/tcpip_adapter.h:86:0: note: this is the location of the previous definition
 #define IP6_ADDR_BLOCK1(ip6addr) ((u16_t)(FreeRTOS_htonl((ip6addr)->addr[0]) >> 16) & 0xffff)
 ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:44:0,
                 from C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:46,
                 from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:21:
C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip6_addr.h:80:0: warning: "IP6_ADDR_BLOCK2" redefined
 #define IP6_ADDR_BLOCK2(ip6addr) ((u16_t)((lwip_htonl((ip6addr)->addr[0])) & 0xffff))
 ^
[... 65 lines cut ...]
                 from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:21:
C:/msys32/home/Benji/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:76:3: error: conflicting types for 'ip_addr_t'
 } ip_addr_t;
   ^
In file included from C:/msys32/home/Benji/esp/esp-idf/components/esp_http_client/lib/http_auth.c:20:0:
C:/amazon-freertos-master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/tcpip_adapter/include/tcpip_adapter.h:62:20: note: previous declaration of 'ip_addr_t' was here
 typedef ip4_addr_t ip_addr_t;
                    ^
cc1.exe: warning: unrecognized command line option '-Wno-int-in-bool-context'
cc1.exe: warning: unrecognized command line option '-Wno-memset-elt-size'
cc1.exe: warning: unrecognized command line option '-Wno-format-truncation'
cc1.exe: warning: unrecognized command line option '-Wno-switch-unreachable'
cc1.exe: warning: unrecognized command line option '-Wno-unused-const-variable'
cc1.exe: warning: unrecognized command line option '-Wno-implicit-fallthrough'
cc1.exe: warning: unrecognized command line option '-Wno-cast-function-type'
cc1.exe: warning: unrecognized command line option '-Wno-misleading-indentation'
cc1.exe: warning: unrecognized command line option '-Wno-stringop-truncation'
cc1.exe: warning: unrecognized command line option '-Wno-format-overflow'
make[1]: *** [/home/Benji/esp/esp-idf/make/component_wrapper.mk:292: lib/http_auth.o] Error 1
make: *** [C:\msys32\home\Benji\esp\esp-idf/make/project.mk:527: component-esp_http_client-build] Error 2

Can anybody help me?

Thanks!
Benji

bhusang, 2019-02-22:
Please refer to the documentation here: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

Especially the part which states:
Note

When you reach the "Get ESP-IDF" instructions under Next Steps, stop and return to the instructions on this page.   If you previously followed the "Get ESP-IDF" instructions and installed ESP-IDF, make sure that you clear the  IDF_PATH environment variable from your system before continuing.

aggarg, 2019-02-22:
Thank you for the pointer @bhusang. Let us know if this fixes your problem.

Thanks.

Benji1992-HU, 2019-02-22:
Thanks, its solved my problem!

closed 2019-02-22 by Benji1992-HU