aws/amazon-freertos #339: Build failed for Nordic Demo

ljaynes87, 2019-02-22:
Nordic build fails on latest commits.

Building ‘aws_demos_ble’ from solution ‘aws_demos_ble’ in configuration ‘Debug’
1> Linking aws_demos_ble.elf
1> Output/aws_demos_ble Debug/Obj/aws_iot_mqtt_serialize.o: In function `AwsIotMqttInternal_InitSerialize':
1> C:\Users\ljaynes\amazon-freertos\lib\mqtt/aws_iot_mqtt_serialize.c:624: undefined reference to `AwsIotMqttBLE_InitSerialize'
Build failed

I can get it to build by renaming some defines and functions in the aws_iot_config.h file.

extern bool AwsIotMqttBLE_InitSerialize( void );
extern void AwsIotMqttBLE_CleanupSerialize( void );
#define AwsIotMqttInternal_InitSerializeAdditional AwsIotMqttBLE_InitSerialize
#define AwsIotMqttInternal_CleanupSerializeAdditional AwsIotMqttBLE_CleanupSerialize 

to

extern bool IotBleMqtt_InitSerialize( void );
extern void IotBleMqtt_CleanupSerialize( void );
#define AwsIotMqttInternal_InitSerializeAdditional IotBleMqtt_InitSerialize
#define AwsIotMqttInternal_CleanupSerializeAdditional IotBleMqtt_CleanupSerialize

This allows me to build the project but I then get an error when turning on the mqtt proxy via the Android SDK.
8 3957 [BLE] Failed to initialize the decoder, error = 7, object type = 4

The error seems to be coming from the static AwsIotSerializerError_t _createDecoderObject() function in  aws_iot_serializer_tinycbor_decoder.c, because  cbor_value_copy_text_string() is returning CborErrorDataTooLarge.

huguesBouvier, 2019-02-23:
Thank you for reporting the problem! We fixed the issue with this PR:https://github.com/aws/amazon-freertos/pull/342

chegewara, 2019-02-24:
Hi,
im having similar issue with esp32. After pairing/bonding im getting those errors:
```
5 78 [OTA] OTA demo version 0.9.2
6 78 [OTA] Creating MQTT Client...
7 78 [OTA] Waiting for a network connection.
8 8042 [Btc_task] [INFO ][DEMO][8042] BLE Connected to remote device, connId = 0

E (82063) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
E (82423) BT_SMP: Value for numeric comparison = 271894
9 8209 [InputTask] Numeric comparison:271894
10 8209 [InputTask] Press 'y' to confirm
11 8756 [InputTask] Key accepted
12 8756 [InputTask] status: 0
W (87913) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
13 10286 [Btc_task] Failed to initialize the decoder, error = 7, object type = 4
14 10295 [Btc_task] Failed to initialize the decoder, error = 0, object type = 3
15 12036 [Btc_task] [INFO ][DEMO][12036] BLE disconnected with remote device, connId = 0
```

PS amazon-freertos is with PR:#342 merged already

ljaynes87, 2019-02-25:
Thanks. It looks like the Nordic build issue was fix. Any idea on the failed to initialize the decoder error?

huguesBouvier, 2019-02-25:
Hello,

We recently updated the protocol from using JSON to CBOR. The advantage is a big boost in speed due to a reduction of overhead. OTA should be faster now.

Those changes entails an update of the APP too. Please download the latest IOS SDK:
https://github.com/aws/amazon-freertos-ble-ios-sdk/
If you are using android, it has not been updated yet but we are working on it and should be updated soon.

I am not closing the ticket, I'll let you know when it is updated and close the ticket.

Hugues

ljaynes87, 2019-03-04:
Looks like the Android SDK was updated and the failed to initialize the decoder error was fixed. Thanks.

huguesBouvier, 2019-03-06:
You are welcome!

closed 2019-03-06 by huguesBouvier