aws/amazon-freertos #345: Espressif ESP32 demo crashes if power management is enabled

thealastair, 2019-02-25:
**Describe the bug**
When power management is enabled (ticking the box "Power Management" from the menuconfig of the ESP32 demo), the unit crashes/reboots constantly.

**System information**
- Which hardware board or part numbers?
ESP-WROVER-KIT

- IDE used
None
- Operating System
Linux

- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
v1.4.6-61-gf0c0efc7
- Project
demos/espressif/esp32_devkitc_esp_wrover_kit

**Expected behavior**
The board should run without rebooting

**Screenshots or console output**
I (31) boot: ESP-IDF v3.1.1-18-g607d2ce34 2nd stage bootloader
I (31) boot: compile time 06:54:47
I (33) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (67) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (75) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (82) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (90) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (97) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (105) boot: End of partition table
E (109) boot: ota data partition invalid and no factory, will try all partitions
I (117) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x14838 ( 84024) map
I (155) esp_image: segment 1: paddr=0x00034860 vaddr=0x3ffc0000 size=0x02bf8 ( 11256) load
I (160) esp_image: segment 2: paddr=0x00037460 vaddr=0x3ffc2bf8 size=0x00000 (     0) load
I (162) esp_image: segment 3: paddr=0x00037468 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/alastairb/contrib/beta-amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

I (171) esp_image: segment 4: paddr=0x00037870 vaddr=0x40080400 size=0x087a0 ( 34720) load
I (194) esp_image: segment 5: paddr=0x00040018 vaddr=0x400d0018 size=0x75ce4 (482532) map
0x400d0018: _stext at ??:?

I (364) esp_image: segment 6: paddr=0x000b5d04 vaddr=0x40088ba0 size=0x08090 ( 32912) load
0x40088ba0: coex_core_release at ??:?

I (378) esp_image: segment 7: paddr=0x000bdd9c vaddr=0x400c0000 size=0x00000 (     0) load
I (378) esp_image: segment 8: paddr=0x000bdda4 vaddr=0x50000000 size=0x00000 (     0) load
I (394) boot: Loaded app from partition at offset 0x20000
I (394) boot: Disabling RNG early entropy source...
I (396) cpu_start: Pro cpu up.
I (400) cpu_start: Single core mode
I (404) heap_init: Initializing. RAM available for dynamic allocation:
I (411) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (417) heap_init: At 3FFD12F0 len 0000ED10 (59 KiB): DRAM
I (423) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (430) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (436) heap_init: At 40090C30 len 0000F3D0 (60 KiB): IRAM
I (442) cpu_start: Pro cpu start user code
I (124) cpu_start: Starting scheduler on PRO CPU.
I (133) wifi: wifi driver task: 3ffd97c4, prio:23, stack:3584, core=0
I (133) wifi: wifi firmware version: 8f35e40
I (133) wifi: config NVS flash: enabled
I (143) wifi: config nano formating: disabled
I (143) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (153) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (173) wifi: Init dynamic tx buffer num: 32
I (173) wifi: Init data frame dynamic rx buffer num: 5
I (173) wifi: Init management frame dynamic rx buffer num: 5
I (173) wifi: Init static rx buffer size: 1600
I (183) wifi: Init static rx buffer num: 5
I (183) wifi: Init dynamic rx buffer num: 5
0 6 [main] WiFi module initialized. Connecting to AP Paste Wi-Fi SSID here....
I (253) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (253) wifi: mode : sta (3c:71:bf:9a:c0:b0)
I (263) WIFI: SYSTEM_EVENT_STA_START
core=0 port_interruptNesting=1

/home/alastairb/contrib/beta-amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:315 (vPortAssertIfInISR)- assert failed!
abort() was called at PC 0x4008ae03 on core 0
0x4008ae03: vPortAssertIfInISR at /home/alastairb/contrib/beta-amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

Backtrace: 0x4008e6b8:0x3ffc0520 0x4008e887:0x3ffc0540 0x4008ae03:0x3ffc0560 0x4008c432:0x3ffc0580 0x40081eea:0x3ffc05a0 0x4008135f:0x3ffc05c0 0x400813f3:0x3ffc05e0 0x4008229c:0x3ffc0600 0x400ea159:0x00000000
0x4008e6b8: invoke_abort at /home/alastairb/contrib/beta-amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x4008e887: abort at /home/alastairb/contrib/beta-amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x4008ae03: vPortAssertIfInISR at /home/alastairb/contrib/beta-amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

0x4008c432: vTaskEnterCritical at /home/alastairb/contrib/beta-amazon-freertos/lib/FreeRTOS/tasks.c:4948

0x40081eea: esp_pm_lock_acquire at /home/alastairb/contrib/beta-amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/pm_locks.c:114

0x4008135f: leave_idle at /home/alastairb/contrib/beta-amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/pm_esp32.c:243

0x400813f3: esp_pm_impl_isr_hook at /home/alastairb/contrib/beta-amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/pm_esp32.c:243

0x4008229c: _xt_lowint1 at /home/alastairb/contrib/beta-amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x400ea159: esp_vApplicationIdleHook at /home/alastairb/contrib/beta-amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/freertos_hooks.c:50

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6336
ho 0 tail 12 room 4
load:0x40078000,len:11264
load:0x40080000,len:6008
0x40080000: _iram_start at /home/alastairb/contrib/beta-amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

**To reproduce**
Steps to reproduce the behavior:
1. Issue occurs on boot, no interaction required

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
Modify demos/espressif/esp32_devkitc_esp_wrover_kit/make/sdkconfig as follows:
-CONFIG_PM_ENABLE=
+CONFIG_PM_ENABLE=y
+CONFIG_PM_DFS_INIT_AUTO=
+CONFIG_PM_USE_RTC_TIMER_REF=
+CONFIG_PM_PROFILING=
+CONFIG_PM_TRACE=

```

aggarg, 2019-02-25:
Thank you for reporting this issue. I am able to repro it. I'll update once we have a fix.

Thanks.

thealastair, 2019-03-06:
Hi, has there been any progress on this?

aggarg, 2019-03-06:
Hi @thealastair, this is fixed with the following PR: https://github.com/aws/amazon-freertos/pull/368 

Would you please try using the latest code from master and let us know if it works for you?

Thanks.

thealastair, 2019-03-08:
Works for me (as in, it runs and doesn't crash). Thanks.

closed 2019-03-08 by thealastair