aws/amazon-freertos #346: xTaskCreate with priority 0

chegewara, 2019-02-25:
Hi,
im beginning journey with amazon-freertos and i dont know how is task scheduler working, but in esp32-idf it is not good practice to create task with priority 0.
https://github.com/aws/amazon-freertos/blob/feature/ble-beta/demos/common/ble/aws_ble_numericComparison.c#L74

This code seems to work, but there may be more tasks created with priority 0 and can happen that some wont get time from scheduler.

huguesBouvier, 2019-02-25:
Hello,

With Amazon FreeRTOS, 0 is Task_IDLE priority. This is a valid task priority, thought it should not be hardcoded.

Is there any reference in ESP32-IDF against using priority 0? 

Best,

Hugues

htibosch, 2019-02-26:
> but there may be more tasks created with priority 0 and
> can happen that some wont get time from scheduler.

With these three macros defined as non-zero:

`#define configUSE_PREEMPTION   1`
`#define configUSE_TIME_SLICING   1`
`#define configIDLE_SHOULD_YIELD   1`

it is no problem to have several tasks running at the same (zero) priority.
Even when these tasks never block, they will all get the same amount of CPU time.

The [manual](https://www.freertos.org/a00110.html) can tell you a lot more about the FreeRTOS macros.

Hein

chegewara, 2019-02-26:
> Hello,
> 
> With Amazon FreeRTOS, 0 is Task_IDLE priority. This is a valid task priority, thought it should not be hardcoded.
> 
> Is there any reference in ESP32-IDF against using priority 0?
> 
> Best,
> 
> Hugues

There is no reference i know of, but there were some issues reported i remember and got the main cause was task priority 0. It was also confirmed by espressif team member @projectgus

projectgus, 2019-02-26:
Hi folks,

In ESP-IDF we [modified the FreeRTOS scheduler to add SMP support](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/freertos-smp.html), so there are some additional ESP-IDF-specific restrictions that don't apply to actual FreeRTOS.

I think in this case the root cause is that in SMP mode ESP-IDF currently doesn't guarantee round robinning of equal priority tasks. Having additional tasks at priority 0, without reliable round robinning, can create the situation where either the idle task on one core is starved, or the additional task is starved. So we don't support this in ESP-IDF, at least not when SMP mode is enabled.

None of this is relevant to Amazon FreeRTOS, to the best of my knowledge.

(@Dazza0 please correct me if I got any of that wrong.)

sarenameas, 2019-03-15:
We could possibly make this demo task priority configurable as it is with other demos, to alleviate any concerns.

cobusve, 2019-05-16:
As per @projectgus this should not be relevant to Amazon FreeRTOS so I am going to close this issue.

closed 2019-05-16 by cobusve