aws/amazon-freertos #353: Packet and/or TCP/IP stack corruption

marjon-halter, 2019-02-27:
**Describe the bug**
TCP Echo demo runs on repeat, but after about 1000++ successful packets, an assert happens saying the sent packet is not the same as the received back packet. Getting packet corruption and/or memory corruption.
The failure is not always at the exact same place of execution, but the assert failure is always the same one.
```cpp
/* Compare the transmitted string to the received string. */
configASSERT( strncmp( pcReceivedString, pcTransmittedString, xTransmitted ) == 0 );
```

**System information**
- Which hardware board or part numbers? ESP-WROVER-KIT V4.1 
- IDE used - Eclipse + IDF programming scripts
- Operating System [Linux]
- Version of Amazon FreeRTOS : v1.4.6-69-g3b189e55
- Project [TCP Echo]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
Device runs the TCP Echo demo to a local TCP Echo Server, sleeps, wakes up, repeat.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

```cpp
5943 95004 [Echo0] Received correct string from echo server.
5944 95004 [Echo0] Sending TxRx message number 1658 of length 1659 to echo server
5945 95004 [Echo0] Received correct string from echo server.
5946 95005 [Echo0] Sending TxRx message number 1659 of length 1660 to echo server
5947 95087 [Echo0] Received correct string from echo server.
5948 95087 [Echo0] Sending TxRx message number 1660 of length 1661 to echo server
5949 95187 [Echo0] Received correct string from echo server.
5950 95187 [Echo0] Sending TxRx message number 1661 of length 1662 to echo server
5951 95499 [Echo0] Timed out receiving from echo server
5952 95499 [Echo0] Received correct string from echo server.
5953 95499 [Echo0] Sending TxRx message number 1662 of length 1663 to echo server
/home/user/workspace/amazon-freertos/demos/common/tcp/aws_tcp_echo_client_single_task.c:323 (prvEchoClientTask)- assert failed!
abort() was called at PC 0x400e1003 on core 0
0x400e1003: prvEchoClientTask at /home/user/workspace/amazon-freertos/demos/common/tcp/aws_tcp_echo_client_single_task.c:323 (discriminator 1)

Backtrace: 0x4008dcb0:0x3ffded10 0x4008de7f:0x3ffded30 0x400e1003:0x3ffded50
0x4008dcb0: invoke_abort at /home/user/workspace/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:676

0x4008de7f: abort at /home/user/workspace/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:676

0x400e1003: prvEchoClientTask at /home/user/workspace/amazon-freertos/demos/common/tcp/aws_tcp_echo_client_single_task.c:323 (discriminator 1)
```

**To reproduce**
Steps to reproduce the behavior:

1. Connect to Access point
2. Run TCP Echo demo code.
3. Disconnect from Access point.
4. Sleep 100ms.
5. Wake up.
6. Repeat.

**Code to reproduce the bug**
If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.
https://gist.github.com/marjon-halter/dabd7ec58d0028508e6bf61076f77845

**Additional context**
Add any other context about the problem here.

Thank you!

htibosch, 2019-02-27:
Hi, thanks for taking the effort to report this.

Does it always stop at message number 1662?

To me it sounds like there is data corruption. The TCP code will drop packets that have a wrong checksum.

Yes it is possible that bytes are shifted in such a way that the checksum remains the same. Very unlikely though unless the connection quality is bad.

Would it be possible to find out what string was received and what was expected?
So in stead of calling `configASSERT()`, just check the result of `strncmp()` and print the differences?

marjon-halter, 2019-02-27:
Hi, 
To answer the questions:
1. Yes, line 1662.
2. The string being sent is the one in the TCP Echo Client Single Task demo, whatever that was.
3. I've posted the changes I did to connect to the accesspoint and sleep. But other than that, it should be a straight clone of the current repo (without my specific SSID and password)

htibosch, 2019-02-27:
All right.
I just see that your logging shows a time-out, I overlooked that in my earlier post.

In that case it would be interesting to add some extra logging.

I suspect that at the moment `SOCKETS_Send()` is called, the reception buffer of the socket is not yet empty.

I am attaching 'aws_tcp_echo_client_single_task.c' with some extra logging.

[aws_tcp_echo_client_single_task.c.gz](https://github.com/aws/amazon-freertos/files/2908693/aws_tcp_echo_client_single_task.c.gz)

htibosch, 2019-02-27:
I think that I found the answer to your question.
I could replicate it with the Windows Simulator project and an embedded echo server at the other end.

I started testing with a length of 1500 and higher. These packets are split up into 2 packets. I instructed the remote end to occasionally put a 2-second delay between the 2 replies: part-1 is returned immediately, the rest is delayed.

As a result, `recv()` does not return enough data, and the `strncmp()` will fail! That is the ASSERT that you saw.

My logging now shows:

	941 34209 [Echo0] Timed out receiving from echo server
	942 34209 [Echo0] xTransmitted 1697 xReceivedBytes 1160

The demo would become nicer if `strncmp()` is only called in case the received length it the same as the expected length.

This extra check will avoid that:

`if( xReceivedBytes != xTransmitted )`
`{`
`    /* The received string did not match the transmitted string. */`
`    ulTxRxFailures[ xInstance ]++;`
`    configPRINTF( ( "ERROR: xTransmitted %d xReceivedBytes %d \r\n", (int)xTransmitted, (int)xReceivedBytes ) );`
`    break;`
`}`

Please find attached `aws_tcp_echo_client_single_task.2.c`. You won't see the ASSERT anymore, unless the memory is corrupted

Hein
[aws_tcp_echo_client_single_task.2.c.gz](https://github.com/aws/amazon-freertos/files/2909538/aws_tcp_echo_client_single_task.2.c.gz)

marjon-halter, 2019-02-27:
If it is isolated to the tcp echo client demo, then it is not critical for us. Will you be updating the demo in the repo? Thanks for investigating this.

closed 2019-02-27 by marjon-halter

htibosch, 2019-02-28:
Although closed: yes I submitted a PR and yes it is isolated to the tcp echo client demo.
Hein