aws/amazon-freertos #357: Outbound not sent successfully

temanXMC, 2019-02-28:
Hello,

I am using Infineon XMC4800 IoT Connectivity kit for connecting with AWS cloud. 

For connecting to AWS cloud, i am following exercises or labs available in github. In lab, XMC4800 kit would do temperature reading and then send the reading to cloud. We want to send the data in JSON format. Also, we want the message to contain our Device ID so when the data gets logged somewhere, it can keep a record of where it came from. 

After compile and run project file, we are finding an issue or error message ("Outbound not sent successfully"). And so temperature reading is not shown in IoT Console. i must have missed a step for outbound not sent successfully message. Could you advise ? Tks.

[OutboundNotSentSuccessfully.pdf](https://github.com/aws/amazon-freertos/files/2914454/OutboundNotSentSuccessfully.pdf)

for detailed info on lab is shown in link below.
https://github.com/rpcme/aws-and-infineon-xmc4800-workshop/blob/master/Lab3.md

Best Regards,
xmcTeman

tgsong, 2019-02-28:
Hi,

Thanks for trying out Amazon FreeRTOS and AWS IoT. Just to make sure, have you completed [lab 1 ](https://github.com/rpcme/aws-and-infineon-xmc4800-workshop/blob/master/Lab1.md) and [lab 2](https://github.com/rpcme/aws-and-infineon-xmc4800-workshop/blob/master/Lab2.md) exercises? The first step is to verify that you have correctly configured `aws_clientcredential.h` and `aws_clientcredential_keys.h`. Then check the policy that attached to your thing allows you to publish message. You can find the detailed guide here, https://docs-aws.amazon.com/freertos/latest/userguide/freertos-prereqs.html#get-started-freertos-thing

Thanks

temanXMC, 2019-03-01:
hi,

yes, i completed lab1 and lab 2 exercises successfully. 

temanXMC

tgsong, 2019-03-01:
If you're able to complete lab 2 and send MQTT message to AWS IoT then your configuration should be correct. From your screenshot it seems like you're able to connect to AWS IoT but unable to publish MQTT messages. I suggest you debug the program and step into `MQTT_Agent_Publish` and see where exactly it fails.

closed 2019-04-29 by dcgaws