aws/amazon-freertos #362: Wifi AP mode doesn't return IP address

iamtonny, 2019-03-01:
Hello,
I'm using esp32 chip with [https://github.com/aws/amazon-freertos/blob/master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/main.c](url) demo code. I wanna start the chip in wifi AP mode, so i changed  

> xWifiStatus = WIFI_ConnectAP( &( xNetworkParams ) );  (219 row)

to 

> xWifiStatus= WIFI_StartAP();

And removed code the following code associated with mqtt demos running:
(158-161 rows)
>  vDevModeKeyProvisioning();
>  DEMO_RUNNER_RunDemos();

After running the modified code on the device, i have the following behavior:

> 0 0 [IP-task] prvIPTask started
> I (134) wifi: wifi driver task: 3ffd9634, prio:23, stack:3584, core=0
> I (134) wifi: wifi firmware version: 8f35e40
> I (134) wifi: config NVS flash: enabled
> I (144) wifi: config nano formating: disabled
> I (144) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
> I (154) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
> I (174) wifi: Init dynamic tx buffer num: 32
> I (174) wifi: Init data frame dynamic rx buffer num: 5
> I (174) wifi: Init management frame dynamic rx buffer num: 5
> I (174) wifi: Init static rx buffer size: 1600
> I (184) wifi: Init static rx buffer num: 5
> I (184) wifi: Init dynamic rx buffer num: 5
> 1 6 [main] ConfigureMe2 6 [main] WiFi module initialized. Connecting to AP alice_g...
> I (1244) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
> I (1244) wifi: mode : softAP (24:0a:c4:13:44:f5)
> I (1244) wifi: Init max length of beacon: 752/752
> I (1244) wifi: Init max length of beacon: 752/752
> I (1254) WIFI: SYSTEM_EVENT_AP_START
> 3 112 [main] WiFi AP started...
> I (24194) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (24194) wifi: station: 88:83:22:a8:4d:9e join, AID=1, bgn, 20
> I (24194) WIFI: SYSTEM_EVENT_AP_STACONNECTED
> I (25014) wifi: station: 88:83:22:a8:4d:9e leave, AID = 1
> I (25014) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (25014) WIFI: SYSTEM_EVENT_AP_STADISCONNECTED
> I (39114) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (39114) wifi: station: 88:83:22:a8:4d:9e join, AID=1, bgn, 20
> I (39124) WIFI: SYSTEM_EVENT_AP_STACONNECTED
> I (75204) wifi: station: 88:83:22:a8:4d:9e leave, AID = 1
> I (75204) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (75204) WIFI: SYSTEM_EVENT_AP_STADISCONNECTED
> I (79474) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (79474) wifi: station: 88:83:22:a8:4d:9e join, AID=1, bgn, 20
> I (79474) WIFI: SYSTEM_EVENT_AP_STACONNECTED
> I (115164) wifi: station: 88:83:22:a8:4d:9e leave, AID = 1
> I (115164) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (115174) WIFI: SYSTEM_EVENT_AP_STADISCONNECTED

So, every time I try to connect to the wifi network of the chip, the following happens:

> I (79474) WIFI: SYSTEM_EVENT_AP_STACONNECTED
> I (115164) wifi: station: 88:83:22:a8:4d:9e leave, AID = 1
> I (115164) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (115174) WIFI: SYSTEM_EVENT_AP_STADISCONNECTED
> 

The device is trying to connect, but cannot get ip address.

**Expected behavior:**
I (...) tcpip_adapter: softAP assign IP to station,IP is: 192.168....

***Actual behavior:*
I (...) WIFI: SYSTEM_EVENT_AP_STADISCONNECTED

Any ideas?

Thank you!

tgsong, 2019-03-04:
Hi,

softAP mode requires DHCP server running on device and it's currently not supported in ESP32. We value your feedback and will take them into consideration in prioritizing our support for features on ESP32.

Thanks.

danhorner, 2019-09-09:
Recording a +1 for the same feature on the PIC32MZ port

tjirsch, 2019-09-29:
Is there any update on this ? It is a bit sad to find basic features that worked id idf version to disappear in the AWS version.

tjirsch, 2019-09-29:
at least remove it from the documentation, please.

dcgaws, 2019-12-23:
@tjirsch can you please point me to the documentation that you reference above?

qiutongs, 2020-02-11:
LwIP support for ESP32 has been merged in https://github.com/aws/amazon-freertos/pull/1519
We are doing further testing on its functionality and make sure it works as expected. After that, we will try testing the DHCP server.

Thanks for requesting this feature.

closed 2020-02-11 by qiutongs