aws/amazon-freertos #365: pxGetNetworkBufferWithDescriptor() returns NULL buffer

alager, 2019-03-05:
**Describe the bug**
Sometimes `pxGetNetworkBufferWithDescriptor()` will return a valid descriptor, but the buffer pointer within is NULL.  All the callers assume that if the descriptor is not NULL then the buffer is also not NULL.

**System information**
- Which hardware board or part numbers?
LPC1788
- IDE used
Crossworks

- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
FreeRTOS 10.2.0

- Project [Custom Application Simple TCP Server]
Just a project that responds to ping

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
If the buffer is NULL, then the descriptor should be released and then return NULL.
```if( pxReturn->pucEthernetBuffer == NULL )
{
     // do something safe with pxReturn
    // return NULL
}
```

**To reproduce**
Steps to reproduce the behavior:
1. create ping project
2. send continuous pings `ping -t IP_ADDRESS`
3. send https request to IP_ADDRESS
4. watch assert happen on line 842 of FreeRTOS_IP.c
`memcpy( pxNewBuffer->pucEthernetBuffer, pxNetworkBuffer->pucEthernetBuffer, pxNetworkBuffer->xDataLength );`

**Code to reproduce the bug**
```cpp
NetworkBufferDescriptor_t *pxGetNetworkBufferWithDescriptor( size_t xRequestedSizeBytes, TickType_t xBlockTimeTicks )
{
NetworkBufferDescriptor_t *pxReturn = NULL;
BaseType_t xInvalid = pdFALSE;
UBaseType_t uxCount;

	/* The current implementation only has a single size memory block, so
	the requested size parameter is not used (yet). */
	( void ) xRequestedSizeBytes;

	if( xNetworkBufferSemaphore != NULL )
	{
		/* If there is a semaphore available, there is a network buffer
		available. */
		if( xSemaphoreTake( xNetworkBufferSemaphore, xBlockTimeTicks ) == pdPASS )
		{
			/* Protect the structure as they are accessed from tasks and
			interrupts. */
			ipconfigBUFFER_ALLOC_LOCK();
			{
				pxReturn = ( NetworkBufferDescriptor_t * ) listGET_OWNER_OF_HEAD_ENTRY( &xFreeBuffersList );

				if( ( bIsValidNetworkDescriptor( pxReturn ) != pdFALSE_UNSIGNED ) &&
					listIS_CONTAINED_WITHIN( &xFreeBuffersList, &( pxReturn->xBufferListItem ) ) )
				{
					uxListRemove( &( pxReturn->xBufferListItem ) );
				}
				else
				{
					xInvalid = pdTRUE;
				}
			}
			ipconfigBUFFER_ALLOC_UNLOCK();

			if( xInvalid == pdTRUE )
			{
				/* _RB_ Can printf() be called from an interrupt?  (comment
				above says this can be called from an interrupt too) */
				/* _HT_ The function shall not be called from an ISR. Comment
				was indeed misleading. Hopefully clear now?
				So the printf()is OK here. */
				FreeRTOS_debug_printf( ( "pxGetNetworkBufferWithDescriptor: INVALID BUFFER: %p (valid %lu)\n",
					pxReturn, bIsValidNetworkDescriptor( pxReturn ) ) );
				pxReturn = NULL;
			}
			else
			{
				/* Reading UBaseType_t, no critical section needed. */
				uxCount = listCURRENT_LIST_LENGTH( &xFreeBuffersList );

				/* For stats, latch the lowest number of network buffers since
				booting. */
				if( uxMinimumFreeNetworkBuffers > uxCount )
				{
					uxMinimumFreeNetworkBuffers = uxCount;
				}

				pxReturn->xDataLength = xRequestedSizeBytes;

				#if( ipconfigTCP_IP_SANITY != 0 )
				{
					prvShowWarnings();
				}
				#endif /* ipconfigTCP_IP_SANITY */

				#if( ipconfigUSE_LINKED_RX_MESSAGES != 0 )
				{
					/* make sure the buffer is not linked */
					pxReturn->pxNextBuffer = NULL;
				}
				#endif /* ipconfigUSE_LINKED_RX_MESSAGES */

				if( xTCPWindowLoggingLevel > 3 )
				{
					FreeRTOS_debug_printf( ( "BUF_GET[%ld]: %p (%p)\n",
						bIsValidNetworkDescriptor( pxReturn ),
						pxReturn, pxReturn->pucEthernetBuffer ) );
				}
			}
			if( pxReturn->pucEthernetBuffer == NULL )
			{
                                // *** **This happens on occasion, but not sure what to do about it?** *** //
				iptraceFAILED_TO_OBTAIN_NETWORK_BUFFER();
                                FreeRTOS_debug_printf( ("ERROR getting pucEthernetBuffer\n" ) );

			}
			else
			{
				iptraceNETWORK_BUFFER_OBTAINED( pxReturn );
			}
		}
		else
		{
			iptraceFAILED_TO_OBTAIN_NETWORK_BUFFER();
		}
	}

	return pxReturn;
}
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

htibosch, 2019-03-06:
Our messages just crossed. The above is not a bug, but a misunderstanding about how BufferAllocation_1.c works. I explained it here:

https://sourceforge.net/p/freertos/discussion/382005/thread/890f0895e9/?limit=250#0f73

When your platform has little RAM, I would recommend using BufferAllocation_2.c and things will work as expected.

alager, 2019-03-06:
For the LPC17xx platform, the project is incomplete for TCP to work, regardless of which buffer scheme is selected.  I chose to go with fixed, as I could track them easier.

Here is what I got working:
In `xNetworkInterfaceOutput()` :
```cpp
...
/* Will the data fit in the Tx buffer? */
if( pxNetworkBuffer->xDataLength < EMAC_ETH_MAX_FLEN ) /*_RB_ The size needs to come from FreeRTOSIPConfig.h. */
{
	/* format the data for the Tx DMA. */
	TXBuffer.ulDataLen = pxNetworkBuffer->xDataLength;
	TXBuffer.pbDataBuf = (uint32_t *)pxNetworkBuffer->pucEthernetBuffer;

        /* track the DMA buffer pointer for release later (in the ISR) */
	trackNetworkBuffers[ trackHead++ ] = pxNetworkBuffer;
	if( trackHead >= ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS )
	{
		trackHead = 0;
	}

	/* copy the buffer to the Tx DMA descriptor. */
	EMAC_WritePacketBuffer( &TXBuffer );
	EMAC_UpdateTxProduceIndex();

        /* Do not release it until it has been sent. */
	xReleaseAfterSend = pdFALSE;

	iptraceNETWORK_INTERFACE_TRANSMIT();

	/* The Tx has been initiated. */
	xReturn = pdPASS;
}
...
```

added to the `ENET_IRQHandler()`
```cpp
...
if( ( ulInterruptCause & EMAC_INT_TX_DONE ) != 0UL )
{
	/* Check if DMA packets have been delivered. */
	vClearTXBuffers();
}
```

Then added the clear function
```cpp
// called from ISR
// release the buffer after the DMA transmitted it.
static void vClearTXBuffers( void )
{
	NetworkBufferDescriptor_t *pxBuffer = trackNetworkBuffers[ trackTail++ ];
	if( trackTail >= ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS )
	{
		trackTail = 0;
	}

    configASSERT( pxBuffer != NULL );
	vNetworkBufferReleaseFromISR( pxBuffer );
}
```

and removed the line that disabled TX interrupts in `xNetworkInterfaceInitialise()` after `EMAC_Init()` enabled both the RX and TX interrupt.

alager, 2019-03-07:
I'll close this and open a new more specific issue on the stack behavior(s) that I'm experiencing.

closed 2019-03-07 by alager

ulyssesfonseca, 2019-12-10:
Anyone have a code that works with LPC1788?