aws/amazon-freertos #366: typo in FreeRTOS_TCP_WIN.c

alager, 2019-03-06:
**Describe the bug**
in `FreeRTOS_TCP_WIN.c`
line 295: `pxNewListItem->pvContainer = ( void * ) pxList;`
should be: `pxNewListItem->pxContainer = ( void * ) pxList;`

p**v**Container does not exist, but p**x**Container does

**System information**
FreeRTOSPlus 10.2.0

**Expected behavior**
should be: `pxNewListItem->pxContainer = ( void * ) pxList;`

**Screenshots or console output**
'ListItem_t {aka struct xLIST_ITEM}' has no member named 'pvContainer'; did you mean 'pxContainer'?

**To reproduce**
Steps to reproduce the behavior:
build a project that includes FreeRTOS_TCP_WIN.c

Thank you!

htibosch, 2019-03-06:
Yes indeed, I noticed that too.

The pointer to the container of a list item has changed type, from `void *` to `struct xLIST *`:

`-    void * configLIST_VOLATILE pvContainer;`
`+    struct xLIST * configLIST_VOLATILE pxContainer;`

Modules are not supposed to get direct access to struct members. In stead, this access macro could be used:

`	listLIST_ITEM_CONTAINER( pxNewListItem ) = ( struct xLIST * configLIST_VOLATILE ) pxList;`

aggarg, 2019-03-06:
You need to set configENABLE_BACKWARD_COMPATIBILITY to 1 in your FreeRTOSConfig.h to fix this.

```
#define configENABLE_BACKWARD_COMPATIBILITY  1
```
Are you talking about a specifc project in our repo or some of your project?

Thanks.

alager, 2019-03-06:
This is just in the latest release.  Nothing specific to my project.  You are correct, that the `configENABLE_BACKWARD_COMPATIBILITY` flag does fix this.  
It seems odd to require that flag for a standard build, not using "old" files.  Maybe I misunderstand the use of the flag?

aggarg, 2019-03-06:
Yes, you are right. As @htibosch explained, it is needed because ```FreeRTOS_TCP_WIN.c``` accesses list members directly instead of using the accessor macro. We will fix it.

Regarding the failure you noticed, 

>> build a project that includes FreeRTOS_TCP_WIN.c

Which project are you building?

Thanks,
Gaurav

alager, 2019-03-06:
@aggarg I'm not completely sure how to answer that.  It's for the NXP LPC1788 chip, but there isn't a complete demo for that, so we've had to roll our own project.  But the file in question is in `FreeRTOSv10.2.0\FreeRTOS-Plus\Source\FreeRTOS-Plus-TCP` which is common to all the projects with TCP.

aggarg, 2019-03-06:
Thanks @alager. That answers my question. I just wanted to make sure that any project in our repo is not broken (i.e. does not have configENABLE_BACKWARD_COMPATIBILITY set to 1).  

Thank you once again.

closed 2019-03-07 by aggarg