aws/amazon-freertos #367: How to create a new ESP32 project with my own app_main?

jrr, 2019-03-06:
**Describe the bug**

The [getting started guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html) covers building and running the included `esp32_devkitc_esp_wrover_kit` ("aws_demos") demo. This works great!

For a standard (non-AFR) ESP-IDF project, you can build it anywhere so long as the `$IDF_PATH` environment variable is configured. In the [documentation](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#start-a-project), an example is copied out of IDF and built in place. This works:

```
$ export IDF_PATH=~/esp/esp-idf
$ cp -R $IDF_PATH/examples/get-started/hello_world .
$ cd hello_world/
$ make all -j
```

I can do something similar with the aws_demos demo:

```
$ export IDF_PATH=~/repos/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf
$ export AMAZON_FREERTOS_PATH=~/repos/amazon-freertos
$ cp -R ~/repos/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make new-project
$ cd new-project
$ make all -j
```

The problem arises when I try to add a component with my own `app_main()`. Here I copy it from the original esp-idf example:

```
$ cp -R ~/esp/esp-idf/examples/get-started/hello_world/main .
$ make all
CC build/main/hello_world_main.o
AR build/main/libmain.a
LD build/aws_demos.elf
~/new-project/build/amazon-freertos-common/libamazon-freertos-common.a(main.o): In function `app_main':
~/repos/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/main.c:136: multiple definition of `app_main'
~/new-project/build/main/libmain.a(hello_world_main.o):/Users/jrr/new-project/main/hello_world_main.c:17: first defined here
collect2: error: ld returned 1 exit status
make: *** [~/new-project/build/aws_demos.elf] Error 1
```

This is because the `amazon-freertos-common` component includes its own main.c:
https://github.com/aws/amazon-freertos/blob/ab5e7d5d2c20b7dd2103dc53b4b1f398ac853132/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/amazon-freertos-common/component.mk#L40

Are we meant to use this component? How can I exclude its app_main?

Where do I go from here? I can assemble a repro repo if it would be helpful, but I'm hoping I'm missing something obvious.

**System information**
- Which hardware board or part numbers? **ESP32**
- IDE used **N/A**
- Operating System **MacOS**
- Version of Amazon FreeRTOS **v1.4.7-18-gab5e7d5d**
- Project **Custom Application**
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`.

gordonwang0, 2019-03-06:
Hi @jrr,

In [component.mk](https://github.com/aws/amazon-freertos/blob/master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/amazon-freertos-common/component.mk#L33), you can add `$(AMAZON_FREERTOS_DEMOS_DIR)espressif/esp32_devkitc_esp_wrover_kit/common/application_code/main.c` to `COMPONENT_OBJEXCLUDE` to exclude the aws_demos `app_main` from the linker.  This should allow you to link your own `app_main`.

jrr, 2019-03-06:
@gordonwang0 in _my_ component.mk, or in the file you linked?

(do I need to patch amazon-freertos to be able to use it?)

gordonwang0, 2019-03-06:
Hi @jrr,

The file I linked should be modified.  You shouldn't have to modify anything else.

jrr, 2019-03-07:
Alright, `COMPONENT_OBJEXCLUDE` worked, but I had to reference the file just as `main.o`, rather than with the longer path.

Is there a reason main.c is in the component? It seems like it should be referenced from [demos/espressif/esp32_devkitc_esp_wrover_kit/make/Makefile](https://github.com/aws/amazon-freertos/blob/master/demos/espressif/esp32_devkitc_esp_wrover_kit/make/Makefile) instead.

Anyone that doesn't want to build their project on a fork of this repo is going to run in to this.

gordonwang0, 2019-03-08:
Hi @jrr,

I don't know the reason for main being in component.mk.  I'll bring up the issue you encountered if we consider restructuring the makefiles.

Please let me know if you have any further questions; otherwise, I'll close this issue.

jrr, 2019-03-09:
@gordonwang0 Would you accept a PR to correct it? It's currently the only thing preventing us from consuming amazon-freertos as a clean submodule.

Thanks for your help.

gordonwang0, 2019-03-11:
@jrr,

Apologies, but currently we aren't able to accept external pull requests on this repo.

mluis, 2019-04-12:
I'm also having the same issue. It seems difficult to create a new clean project based on amazon-freertos.

Is there any detailed information on how to perform this?

It would be cool to use AWS FreeRTOS but with the lack of proper documentation it is just too expensive (as in time-consuming) to start with it.

matteoscordino, 2019-06-25:
after the release of 201906.00, most of what is in this thread is obsolete, but the root issue (no documentation on how to integrate a project with the AFR build system) is still there, just in a new CMake sauce.
see issue #832 for the new version of this =)

sarenameas, 2019-07-03:
Given that #832 is a new version of this issue, I am going to close this. 
Also https://github.com/aws/amazon-freertos/pull/887 has helped to alleviate this.

closed 2019-07-03 by sarenameas