aws/amazon-freertos #375: Compiling demo ESP32 issues

Smeedy, 2019-03-11:
Hi there,

I'm trying to setup a Greengrass/ThingsGraph from the demo directory using some left-over ESP32 devkits. I tried to compile the demo on IDF release/v3.3 and I ran into some compile issues.

`/opt/esp/esp-idf/components/esp_event/default_event_loop.c:67:28: error: 'ESP_TASKD_EVENT_STACK' undeclared (first use in this function)`

I can quick and dirty fix this in the IDF, but than I ran into some other lwip conflicting types and so on. I must be doing something wrong. Could someone share some pointers?

platform: OSX
python: 2.7.14 virtualenv
xtensa: crosstool-ng-1.22.0-80-g6c4433a

Thx,
Martijn

gordonwang0, 2019-03-11:
Hi @Smeedy,

Are you using a separate install of ESP IDF?  Currently, we only support building with the ESP-IDF hosted on this repository.  Please try clearing the IDF_PATH environment variable and then rebuilding.

jghankins, 2019-03-24:
I had the same problem, having made the same mistake.  Missed the statement in the docs about this. Easy to do.  Works like a charm after backing out that IDF_PATH, and rebuilding.

qiutongs, 2019-03-25:
Yeah, there is a note about "ESP-IDF" in ESP starting guide.
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

closed 2019-03-25 by qiutongs