aws/amazon-freertos #377: Method to determine if OTA task has checked for a queued update

thealastair, 2019-03-11:
**Is your feature request related to a problem? Please describe.**  
We are using OTA with a battery powered device which only connects periodically to Wifi. We would like to connect, check for a pending OTA update, perform the update if a job is queued or disconnect as soon as possible if there is no job queued. Currently the OTA_Agent.h interface does not provide a means to determine if the check for a queued job has completed.

The existing OTA_CheckForUpdate function sounds like it should do this, but it seems to just publish a message, not actually check the result?

**Describe the solution you would like.**  
A callback event in OTA_JobEvent_t to indicate the a check has copmleted or fix the implementation of OTA_CheckForUpdate so that it only returns on response from the server and actually indicates whether there is a job or not.

**Additional context**  

Thank you!

gordonwang0, 2019-03-11:
Hi @thealastair,

Many customers have made this feature request -- we're working on improving the OTA update agent to accommodate it.

thealastair, 2019-03-11:
Hi Gordon, thanks for the reply, is there also a new OTA interface coming which uses the new MQTT client? In the feature/ble-beta branch I see there is an implementation of the existing OTA_Agent.h interface using the new client, but I thought there might a change to the interface coming as well?

lt72, 2019-03-12:
Hi @thealastair, we may be changing the interface ahead of publishing the ble-beta work, which uses the latest libraries, but we are not sure yet. Is there any particular reason why you may want to use the new MQTT lib with OTA? I am interested in understanding your requirements fully, as we may re-design OTA a bit. So far I have:
- Ability to check for pending updates (notification) 
- ...

thank you for your feedback!

thealastair, 2019-03-12:
We're using the AWS shadow client as well as OTA and currently this requires two MQTT connections (because the shadow client implementation doesn't allow sharing) which causes resource problems on our platform, so we'd like to migrate to the new IOT framework which I believe supports shared connections.

pvyawaha, 2019-07-26:
Hi Alastair,

Is this issue resolved or are you still facing any problems?

thealastair, 2019-07-28:
There's doesn't appear to have been any change to the API (at least in the master branch), so no, this issue is not resolved.

shreyasbharath, 2019-11-26:
Hi, has there been any progress made on this?

pvyawaha, 2019-12-24:
Hi,

We are working on providing an example for this and will update here when it is merged in master branch.

sameerdasarwad, 2020-01-25:
hi @pvyawaha  is there any update on an example because we are also facing same problem as @thealastair

qiutongs, 2020-02-08:
We are still actively working on this and will merge the fix to master in coming releases.

closed 2020-02-10 by qiutongs

thealastair, 2020-08-07:
@qiutongs This doesn't yet seem to be supported, why was the issue closed? If it is supported, is there an example of how to implement this?

bcprox, 2021-02-15:
@pvyawaha or @qiutongs is there any news on this feature?
Are there any examples of doing OTA while operating on battery power?