aws/amazon-freertos #378: Nordic nrf52840 mqtt-ble FreeRTOS

Miafiaj, 2019-03-12:
Hi,
I've downloaded the feature/ble-beta branch for the mqtt-ble example of AWS documentation with the Nordic nrf52840 card.
The ble connection is working and I can pair the card with my phone using the android-sdk application. The problem is at the start of the connection I get a "[BLE] UNKNOWN BLE EVENT 35" and  when I start the MQTT Proxy from my phone : the programm on the card stops at 0x978.
My case look like https://github.com/aws/amazon-freertos/issues/297, I tried to delete the cache from my bluetooth application, forget the device  and  reboot my phone but it's not working.

I'm using a nrf52840-DK card with a LG G6 phone (Android 8.0 and Bluetooth 4.2).
I use Segger Embedded Studio on Windows 10.

Here the output of the debugging console : 
"The RAM must start at 200046f8, it's size must be 3b908
0 0 [Tmr Svc] [INFO ][MQTT][0] MQTT library successfully initialized.
1 2 [MQTTEcho] [INFO ][DEMO][2] Waiting for a network connection.
2 795 [BLE] Requesting to update ATT MTU to 247 bytes on connection 0x0.3 795 [BLE] sd_ble_gattc_exchange_mtu_request() on connection 0x0 returned busy, will retry.4 795 [BLE] [INFO ][DEMO][795] BLE Connected to remote device, connId = 0

5 859 [BLE] UNKNOWN BLE EVENT 35
6 1114 [InputTask] Numeric comparison:431898
7 1114 [InputTask] Press 'y' to confirm
8 1114 [InputTask] Key accepted
9 1198 [BLE] BLE_GAP_EVT_AUTH_STATUS
10 5795 [MQTTEcho] [INFO ][MQTT][5795] Establishing new MQTT connection.
11 5795 [MQTTEcho] [INFO ][MQTT][5795] Waiting for operation CONNECT to complete."

And here a screenshot when the program stops during the debugging :
[image]

Thank you!

huguesBouvier, 2019-03-13:
Hello,

I didn't manage to reproduce the issue on the latest BLE-BETA branch. Since ble-beta is in constant development, it is possible that you downloaded the branch at a time it was broken.
The commit a the top of the version I checked was: ba89bab8b84d7e9375df6a5b7bb5724a283ba54d.

Another potential issue could be that your app is not up to date with the branch. We recently changed the code to use CBOR instead of JSON. The device and the APP need to be updated, it could be the issue you are seeing.

Best,

Hugues

Miafiaj, 2019-03-14:
Hi,

Thank you for your answer.
I tested with the new version and the program doesn't crash anymore but now I get this errors when I enable the MQTT Proxy : 

"The RAM must start at 200046f8, it's size must be 3b908
0 0 [Tmr Svc] [INFO ][MQTT][0] MQTT library successfully initialized.
1 2 [MQTTEcho] [INFO ][DEMO][2] Waiting for a network connection.
2 460 [BLE] Requesting to update ATT MTU to 247 bytes on connection 0x0.3 460 [BLE] sd_ble_gattc_exchange_mtu_request() on connection 0x0 returned busy, will retry.4 460 [BLE] [INFO ][DEMO][460] BLE Connected to remote device, connId = 0

5 460 [BLE] [INFO ][TASKPOOL][460] Growing a Task pool with a new worker thread...
6 460 [MQTTEcho] Failed to connect to network, 100 retries left
7 525 [BLE] UNKNOWN BLE EVENT 35
8 960 [MQTTEcho] Failed to connect to network, 99 retries left
9 1460 [MQTTEcho] Failed to connect to network, 98 retries left
10 1960 [MQTTEcho] Failed to connect to network, 97 retries left
11 2460 [MQTTEcho] Failed to connect to network, 96 retries left
12 2960 [MQTTEcho] [INFO ][MQTT][2960] Establishing new MQTT connection.
13 2960 [MQTTEcho] [INFO ][MQTT][2960] (MQTT connection 20013a70, CONNECT operation 20012ee8) Waiti14 3460 [MQTTEcho] [WARN ][TASKPOOL][3460] Attempt to cancel a job that is already executing, or ca15 3460 [MQTTEcho] [INFO ][MQTT][3460] (MQTT connection 20013a70, CONNECT operation 20012ee8) Wait 16 3460 [MQTTEcho] [ERROR][MQTT][3460] Failed to establish new MQTT connection, error TIMEOUT.
17 3460 [MQTTEcho] [INFO ][MQTT][3460] Network connection closed on error.
18 3460 [MQTTEcho] [ERROR][DEMO][3460] Failed to create an MQTT connection."

With the lines of error 13 to 16 overlapping each others.
Do you know where this can come from ?

Best regards

huguesBouvier, 2019-03-14:
Hello,

Overlapping is probably message being truncated before the end so the '\n' is not integrated.
The log message limit is 100 character but can be extended as part of the log configuration.
#define configLOGGING_MAX_MESSAGE_LENGTH            100
in FreeRTOSConfig.h

Seeing at the error you are getting I think the APP fails to connect to the broker. 
From those log, BLE seems to be working properly:
- You can connect
- You can pair and Bond
- It reads the MTU, advertise the correct UUID
- It is able to toggle the enable MQTT button.

The issue is most likely in:
1. The setup of the cognito account: https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html

2. The setup of the device in the cloud: If you use another platform, like windows simulator, you could check the credential over ethernet.
Best,

Hugues

Miafiaj, 2019-03-15:
Hi Hugues,

You were right, the problem was coming from one of the IAM roles (I didn't attach the necessary policy to it).
Thank you very much for all your help.
Best regards

closed 2019-03-15 by Miafiaj