aws/amazon-freertos #389: FreeRTOS-Plus-TCP: Memory leak in NetworkInterface.c in WinPCap port

svkers, 2019-03-15:
**Description**
The WinPCap port of the NetworkInterface uses pcap_compile() to create a filter program. After setting the filter program with pcap_setfilter() the compiled filter program is not freed. This leads to a memory leak.

The man page for pcap_freecode() states the following:
> pcap_freecode() is used to free up allocated memory pointed to by a bpf_program struct generated by pcap_compile(3PCAP) when that BPF program is no longer needed, for example after it has been made the filter program for a pcap structure by a call to pcap_setfilter(3PCAP). 

**Valgrind output**
==2934== 128 bytes in 1 blocks are definitely lost in loss record 1 of 1
==2934==    at 0x4C2DBC5: calloc (vg_replace_malloc.c:711)
==2934==    by 0x5059ED1: ??? (in /usr/lib/x86_64-linux-gnu/libpcap.so.1.8.1)
==2934==    by 0x504D0E2: pcap_compile (in /usr/lib/x86_64-linux-gnu/libpcap.so.1.8.1)

**Fix**
Call pcap_freecode() after setting the filter in amazon-freertos/lib/FreeRTOS-Plus-TCP/source/portable/NetworkInterface/WinPCap/NetworkInterface.c, lines 380-390:
```
if( pcap_compile( pxOpenedInterfaceHandle, &xFilterCode, cErrorBuffer, 1, ulNetMask ) < 0 )
{
	printf( "\nThe packet filter string is invalid\n" );
}
else
{
	if( pcap_setfilter( pxOpenedInterfaceHandle, &xFilterCode ) < 0 )
	{
		printf( "\nAn error occurred setting the packet filter.\n" );
	}

        /* free the bpf filter code that has been generated by pcap_compile() */
        pcap_freecode( &xFilterCode );
}
```

sarenameas, 2019-03-15:
Thank you for pointing this out. It is being address.

svkers, 2019-03-16:
The fix (commit 0a262aff1125ddbc16ab297902a464793966e93e) is not quite correct. The memory pointed to by the bpf_program struct parameter now gets only freed if the call to pcap_setfilter() succeeds but it should be freed anyway (i.e. the call to pcap_compile() succeeds):

`
if( pcap_setfilter( pxOpenedInterfaceHandle, &xFilterCode ) < 0 )
{
	printf( "\nAn error occurred setting the packet filter.\n" );
	/* When pcap_compile() succeeds, it allocates memory for the 
	   memory pointed to by the bpf_program struct 
	   parameter.pcap_freecode() will free that memory. */
	pcap_freecode( &xFilterCode );
}
`

For the fix please see the code snippet in my initial issue description. 

Please reopen the issue.

Thank you!

sarenameas, 2019-03-18:
Yes you are right, it should be freed no matter what pcap_setfilter() returns. I was working too hastily. Thanks.

closed 2019-03-18 by sarenameas