aws/amazon-freertos #395: BLE

chegewara, 2019-03-18:
Hi,
i am working on ble features to add ble scan and ble client functionality for esp32. Basically i have it running and almost ready to make initial PR, just i have some question to maintainers who maybe are working on it. For ble scan i need some BLE structure similar to this one in esp-idf, but i wish it would be more generic for all SoC chips:

```
static esp_ble_scan_params_t* xScan_params = {
    .scan_type = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval = 349,
    .scan_window = 49,
    .scan_duplicate = BLE_SCAN_DUPLICATE_DISABLE
};
```

Any help will be appreciated.

huguesBouvier, 2019-03-20:
Hello,

I am sorry, we don’t accept PR for individual at the moment, thought that may change in the future.

It is very hard to get 1 to 1 mapping between stack. If we make to stack similar to ESP-IDF's then it will diverge from another vendor stack.
You pull closer on one side, it will get further away on the other.
Setting up the parameters is made by using those three calls:
    /**
     *
     * @brief Setup scan filter params.
     *
     * @param[in] xFiltParam
     * @return Returns eBTStatusSuccess on successful call.
     */
    BTStatus_t ( * pxScanFilterParamSetup )( BTGattFiltParamSetup_t xFiltParam );

    /**
     *
     * @brief Configure a scan filter condition.
     *
     * @param[in] ucAdapterIf Adapter interface ID. Returned from BTRegisterBleAdapterCallback_t after calling pxRegisterBleApp.
     * @param[in] ulAction
     * @param[in] ulFiltType
     * @param[in] ulFiltIndex
     * @param[in] ulCompanyId
     * @param[in] ulCompanyIdMask
     * @param[in] pxUuid UUID of the attribute.
     * @param[in] pxUuidMask
     * @param[in] pxBdAddr Address of the Remote device.
     * @param[in] cAddrType
     * @param[in] xDataLen
     * @param[in] pcData
     * @param[in] xMaskLen
     * @param[in] pcMask
     * @return Returns eBTStatusSuccess on successful call.
     */
    BTStatus_t ( * pxScanFilterAddRemove )( uint8_t ucAdapterIf,
                                            uint32_t ulAction,
                                            uint32_t ulFiltType,
                                            uint32_t ulFiltIndex,
                                            uint32_t ulCompanyId,
                                            uint32_t ulCompanyIdMask,
                                            const BTUuid_t * pxUuid,
                                            const BTUuid_t * pxUuidMask,
                                            const BTBdaddr_t * pxBdAddr,
                                            char cAddrType,
                                            size_t xDataLen,
                                            char * pcData,
                                            size_t xMaskLen,
                                            char * pcMask );
    /**
     *
     * @brief Sets the LE scan interval and window in units of N*0.625 msec.
     *
     * @param[in] ucAdapterIf Adapter interface ID. Returned from BTRegisterBleAdapterCallback_t after calling pxRegisterBleApp.
     * @param[in] ulScanInterval
     * @param[in] ulScanWindow
     * @return Returns eBTStatusSuccess on successful call.
     */
    BTStatus_t ( * pxSetScanParameters )( uint8_t ucAdapterIf,
                                          uint32_t ulScanInterval,
                                          uint32_t ulScanWindow );

In those you will find most parameters from the structure you mentioned.
From API standpoint, you don't need to map each and every one of the input.

Best,

Hugues

chegewara, 2019-03-20:
Hi,
yes, i know if i will try to match parameters to esp32 then i will not fit other chips, thats why i asked.
Like i said, i have almost ready ble stack for esp32, but i will keep it in my repository.

Thanks

closed 2019-03-20 by chegewara