aws/amazon-freertos #419: Develop MT7697Hx platform by using gcc compiler

cs3sw, 2019-03-26:
Hi,

I saw MT7697 platform can support IAR, Keil and gcc for development.
In amazon freertos user guide, it use Keil tool for development.  
 Is it possible using gcc compiler to compile amazon freertos with MT7697 platform? 

Thank you!

pvyawaha, 2019-03-27:
Hi , 
Yes it is possible to use gcc compiler and setup will depend on what host platform you are using. You will have to use flash tool for flashing the application from https://labs.mediatek.com/en/download/OwVFVNhv

For Windows
1.	Download MSYS2 from https://www.msys2.org, for Windows 7, 64-bits machine, download the msys2-x86_64-<date code>.exe. Follow the steps on MSYS2 homepage to install it.
2.	After the MSYS2 installation completed, install some needed utilities using ‘pacman –S make git’.
3.	Download GNU ARM Toolchain from https://launchpad.net/gcc-arm-embedded/+download.
a.	Choose this version: 4.8-2014-q3-update.
b.	Download Windows ZIP package.
c.	Extract it to gcc-arm-none-eabi-4_8-2014q3-20140805-win32 under C:\msys2\home\user\.
4.	Verify the installation of MSYS2 and toolchain
a.	Click “MSYS2 MSYS” from its folder “MSYS2 64bit” in “All Programs” menu.
b.	A prompt “#” is shown after it started successfully.
c.	Copy-paste this line into it and press enter: “gcc-arm-none-eabi-4_8-2014q3-20140805-win32/bin/arm-none-eabi-gcc –version”.
d.	A few lines of messages stating it is GCC version “4.8.4 20140725” shall be seen.
5.	Add the toolchain to your program search PATH in MSYS2
a.	Edit C:\msys2\home\user\.bash_profile.
b.	Add this line to the end of it: “PATH=~/gcc-arm-none-eabi-4_8-2014q3-20140805-win32/bin:${PATH}”.
6.	Close the MSYS2 terminal session using ‘exit’ command.
7.	Restart MSYS2 again. 
8.	Invoke GCC by ‘arm-none-eabi-gcc –version’ without any path prefix should succeed.

For Linux - Download and setup GNU ARM Toolchain version 4.8-2014-q3-update

cs3sw, 2019-03-28:
Hi,

Thank you for your reply.
After setup gcc compiler environment, any ideal about using gcc compile amazon freertos?
MT7697Hx project seems just have Keil uvision project, no one for gcc part.
How to make this work?

Thank you.

pvyawaha, 2019-03-28:
Hi,

We have included the gnu folder with makefile to build this and can be built using make command , for example - 

1.	Execute “All Programs” -> “MSYS2 64bit” -> “MSYS2 MSYS”.
2.	Change directory to <BASE_FOLDER>/demos/mtk/mt7697hx-dev-kit/gnu.
3.	Execute “make all” to build code.
4.	Execute “ls out” to check the code has been built and output binary file “aws_demo.bin” is ready to be downloaded.

Thank You

cs3sw, 2019-04-03:
Hi,

Thanks for your support.
We will try it.

Best Regards

abhidixi11, 2019-04-07:
Hello
I'm closing the issue, please feel free to open it again if the solution does not work for you.
Thanks

closed 2019-04-08 by abhidixi11