aws/amazon-freertos #422: What Branch to use for production BLE-enabled ESP32with OTA

bgintz-rsc, 2019-03-26:
**Describe the bug**
What Branch to use for production BLE-enabled ESP32 with working OTA?  

**System information**
- ESP32
- Eclipse ESP/IDF
- MacOS
- Version of Amazon FreeRTOS 
- Custom Application using AFR BLE demo app for provisioning and AFR OTA job

**Expected behavior**
See title.  BLE-beta branch is very active and we need to do a production release ASAP to support OTA of final code in product without access to serial port.

lt72, 2019-03-27:
Hi @bgintz-rsc, you are raising a valid concern. 
Although ble-beta is going to be stable after today, you will still see code changes for minor features you may not need, bug fixes, fit&finish work on demos, fixing projects for boards other than Espressif, etc.
I would propose that we checkout a release candidate branch out of feature/ble-beta by the end of this week specifically for ESP32, and we establish a bar for moving code from ble-beta to such branch.
Depending on your exact release schedule, we will need to be more or less strict about what changes in feature/ble-beta we want to consider moving to the release candidate branch.

huguesBouvier, 2019-03-28:
Hello,

I tested the following commit on feature/ble-beta: 17ba347f1c5cbec9c6f8722542857312f0ee051d.
It is compatible with current latest APP on android and IOS.
https://github.com/aws/amazon-freertos-ble-ios-sdk/tree/beta: 58d66fb81afe3558a7ce711e71dbd58d0d3f0b41
https://github.com/aws/amazon-freertos-ble-android-sdk/tree/beta: ade60b43be9f9eeabe7a2765152b2c111c1a9ab8

You can test using the following:
- following the guide to setup OTA : https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ota-dev.html (create things, OTA job etc...)
- Uncomment the demo: vStartOTAUpdateDemoTask();  https://github.com/aws/amazon-freertos/blob/fda8d37cc1e8acc618b908ad2b0f098c0fe2af0c/demos/common/demo_runner/aws_demo_runner.c#L63
- Flash, run and provision WIFI using the APP.
- Restart
- Run again, OTA should start.

Hugues

bgintz-rsc, 2019-03-28:
Thank you Hughes.  

We are able to get OTA working with this commit.  However we did identify another issue.  

If we attempt to connect to a broker endpoint that is not an "ats" endpoint we can connect successfully and the OTA process begins.

When we use an ATS broker endpoint the connect fails.  

This was resolved by setting  AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER.pRootCA to  tlsATS1_ROOT_CERTIFICATE_PEM

bgintz-rsc, 2019-03-28:
Hughes - 

Could you also please confirm that JITP was tested and works in this commit?  Do we need to do anything new for JITP other than using the JIT generated credentials and initializing:

 clientcredentialJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM?

Thanks!

abhidixi11, 2019-04-02:
Hello @bgintz-rsc ,
>>>When we use an ATS broker endpoint the connect fails.
You have to use ATS  root certificate for connecting to ATS endpoint , so root CA for non ATS and ATS endpoints may be different. (https://docs.aws.amazon.com/iot/latest/developerguide/managing-device-certs.html) , 
>>>This was resolved by setting AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER.pRootCA to >>>tlsATS1_ROOT_CERTIFICATE_PEM
So setting pRootCA to tlsATS1_ROOT_CERTIFICATE_PEM may have helped.   While connecting non ATS endpoint, were you using a different root CA ?
Let us know if you  still questions about the connection.
 
>>>>Could you also please confirm that JITP was tested and works in this commit?
I don't think JITP is tested for ble-beta branch yet. But it should work just as the regular branch.

abhidixi11, 2019-04-02:
`This was resolved by setting AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER.pRootCA to tlsATS1_ROOT_CERTIFICATE_PEM`
It looks like we are not setting pRootCA in our demo.  I think that's what you indicated in the above comment. Non ATS broker endpoint probably works without that, but ATS endpoint doesn't. 
We will update the demo to clarify that before the release, thanks!

bgintz-rsc, 2019-04-03:
This is correct Abhijit.  non-ATS worked with the default RootCA.  ATS required we set it as above.

Thank you!
Bobby

abhidixi11, 2019-04-03:
Thanks for confirming. I'm going to close the issue, feel free to reopen  if it has not answered all of your questions.

closed 2019-04-03 by abhidixi11

bgintz-rsc, 2019-04-03:
Agreed.

Many Thanks
Bobby