aws/amazon-freertos #435: Task Priorities Cannot Be Assigned

gokcedilek, 2019-03-28:
Hello,
I have been having a very weird issue that I don't know what the cause could be. I aimed to create 4 tasks, and later on use ```uxTaskPriorityGet``` and ```vTaskPrioritySet``` in my tasks to switch between them. However, this approach did not work, even though the code and the logic are quite simple. Then, I went back and checked if the tasks are assigned the correct priorities in the beginning; and saw that they are not! My example is below:
```UBaseType_t priority;
      xTaskCreate(exti_line13, "deferred_13", 1000, NULL, 2, &line13Handle);
      xTaskCreate(exti_line10, "deferred_10", 1000, NULL, 1, &line10Handle);
      xTaskCreate(deferred, "deferred_default", 1000, NULL, 4, &deferredHandle);
      xTaskCreate(regular, "regular", 1000, NULL, 3, &regularHandle);
      HAL_UART_Transmit(&huart2, s, sprintf(s, "%d\n", uxTaskPriorityGet(line13Handle)), 1000);
      HAL_UART_Transmit(&huart2, s, sprintf(s, "%d\n", uxTaskPriorityGet(line10Handle)), 1000);
      HAL_UART_Transmit(&huart2, s, sprintf(s, "%d\n", uxTaskPriorityGet(deferredHandle)), 1000);
      HAL_UART_Transmit(&huart2, s, sprintf(s, "%d\n", uxTaskPriorityGet(regularHandle)), 1000);
```
Here, we would expect the priorities to be 2,1,4,3; but they are 2,1,4,4, and this situation depending on how I order the tasks; I tried this many times and I can't see any logical reason why this is happening!
Could you please inform me about this issue?

I appreciate your help, thank you!

RichardBarry, 2019-03-29:
Is this before or after the scheduler is running?

Can you please show the declaration of the task handles line13Handle, line10Handle, deferredHandle and regularHandle and let me know where they are defined (for example, are they file scope variables, global variables, declared on a stack, etc.).  If you inspect each handle, does each have a unique value (asking depending on the answer to the first question)?

If you store the return values of uxTaskPrioityGet() in a variables, rather than calling the function as a parameter to HAL_UART_Transmit(), then inspect the variables in the debugger, are the values still unexpected?

WITH THE SCHEDULER RUNNING, can you please also try calling xTaskCreate(), then immediately calling uxTaskPriorityGet(), to see if the return of uxTaskPriorityGet() is as expected then.

danhorner, 2019-04-23:
Hi @gokcedilek, 

Were you able to resolve this issue? If you've checked out the scope and identity of the task handles as @RichardBarry Richard suggested, can you reopen this ticket and let us know which port you are using?

closed 2019-04-23 by danhorner

gokcedilek, 2019-05-06:
Hello,

Sorry I had to take a break from this project. I would like to ask questions about the suggestions above:
1) How exactly should I inspect the handle values (ie whether they are unique or not)?
2) Should I create the tasks after starting the scheduler? Right now, I was doing the task creation before starting the scheduler (ie before running ```vTaskStartScheduler();```)

The declaration of the handles & tasks:
```
TaskHandle_t line13Handle = NULL;
TaskHandle_t line10Handle = NULL;
TaskHandle_t deferredHandle = NULL;
TaskHandle_t regularHandle = NULL;
TaskHandle_t demoHandle = NULL; //handles are global
xTaskCreate(exti_line13, "deferred_13", 1000, NULL, 1, &line13Handle);
xTaskCreate(exti_line10, "deferred_10", 1000, NULL, 1, &line10Handle);
xTaskCreate(demo, "demo", 1000, NULL, 1, &demoHandle);
xTaskCreate(regular, "regular", 1000, NULL, 2, &regularHandle);
xTaskCreate(deferred, "deferred_default", 1000, NULL, 1, &deferredHandle); //tasks
vTaskStartScheduler(); //scheduler
 ```

gokcedilek, 2019-05-06:
I am using the STM32F401RE microcontroller, and ports A and C.

RichardBarry, 2019-05-06:
There is no problem creating the tasks before the scheduler has started.

As per my first post - where are the handles declared, that is, where does the line "TaskHandle_t line13Handle = NULL;" appear in your code?  Is line13Handle a stack variable?  If so, which stack is it on?  If not stack variables, are they file scope?  Global?  or something else?

If the code you posted all appears inside main(){} then the variables are on the stack used by main().  Once the scheduler starts that stack is recovered for use by interrupt service routines - which means you cannot access any variables stored on the stack used by main() after the scheduler has started.  If that is indeed what you are doing then try moving the declaration of the handles outside of main(){} to prevent them being on the stack.

gokcedilek, 2019-05-07:
Oh I see, that is a good point. However, I declared the handles outside of the main.
- The handles are global.
- The tasks are created inside main(), before starting the scheduler.
When I print the priority values at this point, the values are wrong.
- After the scheduler starts, the task executing functions take place.
However, the functions with the highest priority don't get executed, the lower priority ones do; which tells that the priority values in here are also wrong.

gokcedilek, 2019-05-07:
I just checked and my STM32 application is currently using ```heap_4.c``` as the FreeRTOS heap implementation. Could that be related to the problem that I am experiencing, should I change that to one of the other options?

RichardBarry, 2019-05-07:
> When I print the priority values at this point, the values are wrong.

Are you determining the values are wrong because the values print out incorrectly?  If so, are you absolutely sure the issue is not just that the print function is not working correctly?  Please set a break point in the code and actually read the values using the debugger.

> the functions with the highest priority don't get executed

Are you sure the task actually got created?  Check the return value of the call to xTaskCreate() used to create the task.  If it returns pdPASS, then set a break point at the start of the task to see if it ever gets hit.  If it does get hit, step through the function to see how far it gets.

> I just checked and my STM32 application is currently using heap_4.c

That has nothing to do with task priorities, but if the heap is too small it could prevent a task being created in the first place, as per my previous comment you need to check the return value of xTaskCreate.  You can also use a malloc failed hook to see if a memory allocation fails.  https://www.freertos.org/a00016.html .  Would also suggest looking through this list of common issues https://www.freertos.org/FAQHelp.html 

If you still have issues we will need to see some of the code.