aws/amazon-freertos #452: esp32 FreeRTOS_GetAddressConfiguration returns wrong mac address

chegewara, 2019-03-31:
esp32 is having 4 mac addresses, from which 2 are wifi related. one mac is for AP and one mac is for STA. In STA mode (connected to router AP) `FreeRTOS_GetAddressConfiguration` is returning AP mac address.

abhidixi11, 2019-04-02:
Hello @chegewara,
Can you please  explain what do you mean by "FreeRTOS_GetAddressConfiguration() is returning AP mac address" ?  The function returns IP address/ Gateway address obtained from DHCP server.  Where are you seeing mac address. ? Can you please  post some logs? Thanks.

chegewara, 2019-04-02:
Sorry, my bad, of course i meant `WIFI_GetMAC`.

abhidixi11, 2019-04-04:
Thanks @chegewara ,
As you have pointed out, in aws_wifi.c  WIFI_GetMAC() calls esp_wifi_get_mac with WIFI_MODE_STA (hardcoded here) ,which is equivalent of ESP_IF_WIFI_AP, so AP address is returned instead of STA. 

I'll make sure change is made soon, it should be interface and not mode.  It should be either ESP_IF_WIFI_STA or WIFI_IF_STA as defined in esp_wif_types.h. 
Thanks.

abhidixi11, 2019-04-04:
I have created PR. It will be merged soon. Closing the issue. https://github.com/aws/amazon-freertos/pull/481

closed 2019-04-04 by abhidixi11