aws/amazon-freertos #454: LTE support

Cyrillius, 2019-04-01:
Hello,

I am working on some wearables projects and we are actually using AWS cloud through a mobile application.

We are thinking about using LTE to connect the wearables directly to the cloud whitout going trough a phone.

Do you know if on your roadmap your are planning to support LTE with for example the nRF91?

Best regards,

Cyril

abhidixi11, 2019-04-02:
Hello @Cyrillius ,
Thanks for contacting us.  In future, we may support some cellular technologies, but at this time, I don't have any specifics plans to share.  Please feel free to open the issue again if you have any more questions.

closed 2019-04-02 by abhidixi11

Darsh-Dev, 2020-02-26:
Hey, 

Does the AWS FreeRTOS + LTE support enable for any platform communicate to AWS cloud? 

Regards,
Darshak

aggarg, 2020-02-26:
The FreeRTOS protocol stack includes two application protocols - MQTT and HTTP. Both of these application protocols use a network abstraction layer to talk to the connectivity module (WiFi, Ethernet etc.). To make it more clear, Look at the following diagram:

```
+----------------------+
|      MQTT, HTTP      |
+----------------------+
            |
            v
+----------------------+
|  Network Abstraction |
+----------------------+
            |
            v
+----------------------+
|          TLS         |
+----------------------+
|          TCP         |
+----------------------+
```

FreeRTOS currently does not provide a reference integration for any LTE module but you can write an implementation of Network Abstraction to enable communication with your LTE module - This would enable you to use MQTT to connect to AWS IoT or HTTP to connect to any other AWS service.

Hope that helps.

Thanks.