aws/amazon-freertos #461: LwIP support in ble-beta

caleb-garrett, 2019-04-01:
Is LwIP support considered stable in the feature/ble-beta branch? I am finding that when used with iot_network_afr.c as the network interface LwIP APIs are called from multiple threads, which I do not believe is supported by LwIP and is causing hard faults on my custom target.

Is there a plan to provide a port of iot_network_afr.c for LwIP where SOCKETS_Recv() is not called from a separate thread?

Thanks!

abhidixi11, 2019-04-02:
Hello @caleb-garrett 
iot_network_afr.c provides you two options to receive data from socket:

1. Provide a callback function, where network manager task will poll for you and call your function when data is received.

2. Directly call .receive from IotNetworkInterface_t which will block till receive data (it calls SOCKETS_Recv()). 

You should not use both.  If you don't want to use receive callback, then do not call setReceiveCallback().
Let me know if that explains why you are seeing hard faults.  Otherwise, please let us know what application are you trying to run using  iot_network_afr.c?  It will be helpful know how you are running into this issue. 
If your application does not need to use two network interfaces (BLE and WIFI for example) then you can directly use AFR socket interface as well. 
Thanks.

caleb-garrett, 2019-04-02:
@abhidixi11 Thanks for your quick response. My application is using the MQTT agent, and LwIP is the network stack (needed for PPP) used via the Secure Sockets interface. I only have one network interface.

In the v1.4.7 release, the MQTT agent calls the secure socket interface directly and only from a single thread. In ble-beta it does not look like SOCKETS_Recv() is called from anywhere except _networkReceiveTask, regardless of whether a callback is set, which is a secondary thread. What I am seeing is the Secure Sockets interface is accessed from threads other than _networkReceiveTask simultaneously. For instance, SOCKETS_Shutdown() is called directly from the application thread via IotMqtt_Disconnect() while SOCKETS_Recv() is blocking in _networkReceiveTask. This causes the LwIP mailbox to be deleted, and then later accessed causing a hard fault.

From my understanding of LwIP, I don't believe it is safe to access the Secure Sockets LwIP port from more than one thread, unless careful synchronization is implemented. Does this make sense?

abhidixi11, 2019-04-03:
@caleb-garrett 
Thanks for the details. We are looking it.  We'll update the thread soon.

caleb-garrett, 2019-04-11:
Hi, just wanted to check in to see whether there was any update on this issue. Thanks!

ravibhagavandas, 2019-04-15:
Hi @caleb-garrett ,
Apologize for the delay in response. Unfortunately I cant give you a timeline on when the issue will be fixed. The issue is high in priority in our list of work items, and we will start working on this as soon as we have the bandwidth.

cobusve, 2019-07-08:
Hello @caleb-garrett,

I am currently looking into this issue. 

You did not mention what hardware you are using so I used the MediaTek board which uses LwIP. With this board I tried to reproduce the problem but I was unable to get it to hard fault, it looks like the race you seem to be hitting is very narrow and hard to reproduce.

I see that LwIP does support this use case "Enable code that allows reading from one thread, writing from a 2nd thread and closing from a 3rd thread at the same time." when you have defined LWIP_NETCONN_FULLDUPLEX in the LwIP options for the application as per https://www.nongnu.org/lwip/2_1_x/group__lwip__opts__netconn.html#ga7bfe8487a3abffdd9d6730977d22c406

It does not look like this is defined by default in the code base, so we will have to add that for the ports that use LwIP since we are using it in this way. I will be making this change shortly, in the meantime you can define this and see if that solves your problem. Please do report it here if you still see hard faults with LWIP_NETCONN_FULLDUPLEX = 1 defined in your LwIP options so that we can look into this more deeply to understand exactly why.

caleb-garrett, 2019-07-08:
Hello @cobusve,

I am using custom hardware with the LwIP stack, let me know if you are looking for specific details. Triggering a hard fault is highly dependent on timing, but it is very possible in LwIP when accessing a socket from multiple threads since the stack is not designed to operate this way. From the official LwIP documentation:

> Netconn or Socket API functions are thread safe against the core thread but they are not reentrant at >the control block granularity level. That is, a UDP or TCP control block must not be shared among >multiple threads without proper locking.
>https://www.nongnu.org/lwip/2_1_x/multithreading.html

I chose not to enable LWIP_NETCONN_FULLDUPLEX as a workaround due to the LwIP documentation stating "ATTENTION: This is currently really alpha!" which does not give me the warm and fuzzies. I haven't found any indication anywhere that this feature is stable.

cobusve, 2019-07-09:
I need specifically to know which processor you are compiling for and which development tools you are using (particulalry which compiler). There are some differences from vendor to vendor (in LwIP options and arch ports for one), so if I make a deep analysis of the Cypress implementation and you tell me you are on Mediatek then all my work could be in vain.

In terms of LWIP_NETCONN_FULLDUPLEX not being stable, I guess all we can do is apply subjective judgement here. This was added in 02/2015 so it has been out there for 4 1/2 years so I would say that it should be considered pretty stable by now. Last bug fix that mentioned this feature was in 2015 and I do not see any open issues against this. I also contacted the guys at LwIP and they recommended that we use this and said that this should work. On the other hand of course the documentation still warns about this being Alpha. I will ask the guys there why they are still leaving that comment in the documentation and let you know what they say.

caleb-garrett, 2019-07-10:
@cobusve Thanks for investigating the full duplex issue. I will work on implementing and testing LWIP_NETCONN_FULLDUPLEX on my setup, I look forward to hearing how it goes for you. My understanding is this option is intended to address the issue I have, so as long as it's stable it should do the trick.

I am using an STM32F7 with IAR v8 as compiler. I am using LwIP 2.0.3. Here are the options I have changed from the LwIP default:

`/*----- Default Value for LWIP_DNS: 0 ---*/
#define LWIP_DNS 1
/*----- Default Value for MEM_LIBC_MALLOC: 0 ---*/
#define MEM_LIBC_MALLOC 1
/*----- Default Value for MEMP_MEM_MALLOC: 0 ---*/
#define MEMP_MEM_MALLOC 1
/*----- Value in opt.h for MEM_ALIGNMENT: 1 -----*/
#define MEM_ALIGNMENT 4
/*----- Default Value for MEM_SIZE: 1600 ---*/
#define MEM_SIZE 0
/*----- Value in opt.h for LWIP_ETHERNET: LWIP_ARP || PPPOE_SUPPORT -*/
#define LWIP_ETHERNET 1
/*----- Value in opt.h for LWIP_DNS_SECURE: (LWIP_DNS_SECURE_RAND_XID | LWIP_DNS_SECURE_NO_MULTIPLE_OUTSTANDING | LWIP_DNS_SECURE_RAND_SRC_PORT) -*/
#define LWIP_DNS_SECURE 7
/*----- Value in opt.h for TCP_SND_QUEUELEN: (4*TCP_SND_BUF + (TCP_MSS - 1))/TCP_MSS -----*/
#define TCP_SND_QUEUELEN 9
/*----- Value in opt.h for TCP_SNDLOWAT: LWIP_MIN(LWIP_MAX(((TCP_SND_BUF)/2), (2 * TCP_MSS) + 1), (TCP_SND_BUF) - 1) -*/
#define TCP_SNDLOWAT 1071
/*----- Value in opt.h for TCP_SNDQUEUELOWAT: LWIP_MAX(TCP_SND_QUEUELEN)/2, 5) -*/
#define TCP_SNDQUEUELOWAT 5
/*----- Value in opt.h for TCP_WND_UPDATE_THRESHOLD: LWIP_MIN(TCP_WND/4, TCP_MSS*4) -----*/
#define TCP_WND_UPDATE_THRESHOLD 536
/*----- Default Value for LWIP_NETIF_STATUS_CALLBACK: 0 ---*/
#define LWIP_NETIF_STATUS_CALLBACK 1
/*----- Default Value for LWIP_NETIF_LINK_CALLBACK: 0 ---*/
#define LWIP_NETIF_LINK_CALLBACK 1
/*----- Value in opt.h for TCPIP_THREAD_STACKSIZE: 0 -----*/
#define TCPIP_THREAD_STACKSIZE 1024
/*----- Value in opt.h for TCPIP_MBOX_SIZE: 0 -----*/
#define TCPIP_MBOX_SIZE 6
/*----- Value in opt.h for SLIPIF_THREAD_STACKSIZE: 0 -----*/
#define SLIPIF_THREAD_STACKSIZE 1024
/*----- Value in opt.h for SLIPIF_THREAD_PRIO: 1 -----*/
#define SLIPIF_THREAD_PRIO 3
/*----- Value in opt.h for DEFAULT_THREAD_STACKSIZE: 0 -----*/
#define DEFAULT_THREAD_STACKSIZE 1024
/*----- Value in opt.h for DEFAULT_THREAD_PRIO: 1 -----*/
#define DEFAULT_THREAD_PRIO 3
/*----- Value in opt.h for DEFAULT_UDP_RECVMBOX_SIZE: 0 -----*/
#define DEFAULT_UDP_RECVMBOX_SIZE 6
/*----- Value in opt.h for DEFAULT_TCP_RECVMBOX_SIZE: 0 -----*/
#define DEFAULT_TCP_RECVMBOX_SIZE 6
/*----- Value in opt.h for DEFAULT_ACCEPTMBOX_SIZE: 0 -----*/
#define DEFAULT_ACCEPTMBOX_SIZE 6
/*----- Default Value for LWIP_TCP_KEEPALIVE: 0 ---*/
#define LWIP_TCP_KEEPALIVE 1
/*----- Default Value for LWIP_SO_SNDTIMEO: 0 ---*/
#define LWIP_SO_SNDTIMEO 1
/*----- Default Value for LWIP_SO_RCVTIMEO: 0 ---*/
#define LWIP_SO_RCVTIMEO 1
/*----- Value in opt.h for RECV_BUFSIZE_DEFAULT: INT_MAX -----*/
#define RECV_BUFSIZE_DEFAULT 2000000000
/*----- Default Value for PPP_SUPPORT: 0 ---*/
#define PPP_SUPPORT 1
/*----- Value in opt.h for LWIP_STATS: 1 -----*/
#define LWIP_STATS 0
/*----- Value in opt.h for CHECKSUM_GEN_IP: 1 -----*/
#define CHECKSUM_GEN_IP 0
/*----- Value in opt.h for CHECKSUM_GEN_UDP: 1 -----*/
#define CHECKSUM_GEN_UDP 0
/*----- Value in opt.h for CHECKSUM_GEN_TCP: 1 -----*/
#define CHECKSUM_GEN_TCP 0
/*----- Value in opt.h for CHECKSUM_GEN_ICMP: 1 -----*/
#define CHECKSUM_GEN_ICMP 0
/*----- Value in opt.h for CHECKSUM_GEN_ICMP6: 1 -----*/
#define CHECKSUM_GEN_ICMP6 0
/*----- Value in opt.h for CHECKSUM_CHECK_IP: 1 -----*/
#define CHECKSUM_CHECK_IP 0
/*----- Value in opt.h for CHECKSUM_CHECK_UDP: 1 -----*/
#define CHECKSUM_CHECK_UDP 0
/*----- Value in opt.h for CHECKSUM_CHECK_TCP: 1 -----*/
#define CHECKSUM_CHECK_TCP 0
/*----- Value in opt.h for CHECKSUM_CHECK_ICMP: 1 -----*/
#define CHECKSUM_CHECK_ICMP 0
/*----- Value in opt.h for CHECKSUM_CHECK_ICMP6: 1 -----*/
#define CHECKSUM_CHECK_ICMP6 0`

cobusve, 2019-07-15:
We had some offline communication so I just want to upadte this thread so everyone can benefit.

We have confirmed the behavior that @caleb-garrett  reported here. With LwIP 2.0.3 this will be a problem because this version of LwIP does not support closing a socket from a thread different from the one you are reading the same socket with.

The latest version of LwIP does have an option to support this, that option is enabled by LWIP_NETCONN_FULLDUPLEX in the options file, and you also need to make some changes to the port files for your platform, specifically you need to add code to abort the read on the other thread when the close happens.

This all is obviously not a trivial fix, we will be looking to upgrade LwIP and resolve this as soon as we practically can do this, we will close this issue when this implementation has been completed.

This will only affect you if you are using one of the boards that make use of LwIP, specifically if you are using a board that uses FreeRTOS+TCP you will not be affected by this issue.

mnelson920, 2019-07-24:
Is the ESP32 affected by this?