aws/amazon-freertos #471: FreeRTOS DNS Behavior

sbertrand-witekio, 2019-04-03:
How does the DNS cache uses entries passed aged out records ?

Ageing out of record is done by setting the [DNS name to a NULL string](https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DNS.c#L1367). 

However looking for each entry in the DNS table stop at the [first NULL string DNS Name](https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DNS.c#L1349).

Therefore when a record is aged out, all record of higher index in the table will not be checked.

Furthermore when ageing out the record, the [IP value is not given](https://github.com/aws/amazon-freertos/blob/5b2967a59ecf8b3a1385bca0797ad93842d9bee3/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DNS.c#L1367) but the process is reported as completed (`xFound=pdTRUE`).

abhidixi11, 2019-04-03:
Hello @sbertrand-witekio,

Thanks for pointing it out.  I checked the code looks like a bug.  Did you actually see  DNS cache lookup failing or did you just noticed it in the code? 
I'll verify it again and post an update.
Thanks.

sbertrand-witekio, 2019-04-03:
I just looked that the code, my DNS cache table is only of 1 so that configuration does not have the issue.
I can do a few tests with a bigger cache table.

Another issue I found was the [IP address reported](https://github.com/aws/amazon-freertos/blob/master/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DNS.c#L436) when a record is found, is not formatted correctly, I added a FreeRTOS_ntohl to have the correct byte order.

abhidixi11, 2019-04-08:
Hello @sbertrand-witekio ,
We will work on fixing the issue. 
Thanks

abhidixi11, 2019-04-15:
I have committed the changes and closing the issue.  Thanks.

closed 2019-04-15 by abhidixi11